smartdevicelink/sdl_core #11: Add a default response for startH264

henry-fan, 2015-01-19:
add a default response to notify the mobile app to start encoding video stream.

justinjdickow, 2015-01-27:
@henry-fan Could you please give some context for this issue and what behavior is required?

henry-fan, 2015-02-03:
[image]

lea255ace, 2015-02-03:
The video encoder on the mobile side is on our critical path. The decision to start sending video immediately after the StartH264 message rather than waiting until receiving a response from SDL core serves to hide some of the encoder latency.
In the case of a failed connection, the NACK from SDL core will cause the mobile to close down the connection.

closed 2016-04-12 by justinjdickow