smartdevicelink/sdl_core #13: SDL should notify the applications if the head unit switches between day and night mode

justinjdickow, 2015-01-20:
It should also tell the application about the current mode during registeration

idamaster, 2015-01-21:
We will need to document, for all applications (probably in the soft buttons UI) that day graphics should/may need to be black, and night mode graphics should/may need to be white. Perhaps we add documentation for graphics colors ranges for day night, and provide a format. 

Perhaps we can propose methodology for app developers such as the following:
 Day unselected: icon_day.png
 Day selected: icon_day_swapped.png
 Night unselected: icon_night.png
 Night selected: icon_night_swapped.png

??

tpulatha, 2015-01-22:
For the sake of SDL this probably should be more of an API that reports the background color so that apps can dynamically choose themselves. 
First idea: SDL would return its background color on a scale from 0 .. 255 and the app could decide whether to use the light or the dark icon.

Please note that this is on our list to discuss next week.

JackLivio, 2018-03-01:
Please consider adding this proposal to the sdl evolution repository.

https://github.com/smartdevicelink/sdl_evolution

closed 2018-03-01 by JackLivio