smartdevicelink/sdl_core #16: [Spec] StartServiceACK control frame is not backward compatible

joeljfischer, 2015-01-23:
To have backward compatibility, the sequence of events runs as follows:

1. The mobile library sends a v1 style StartService packet
2. The HU sends a StartServiceACK packet with the highest version it supports
3. The mobile library starts sending packets with the correct negotiated version

The issue I believe could happen is that in (2), the HU doesn't necessarily send a v1 packet for it's StartServiceACK. If it sends a newer style (v2) packet for this control frame, there could be crashes. This would mean that mobile library backcompat is broken from v2+ to v1. It still works from v3 to v2 because the packet structure is the same.

justinjdickow, 2015-02-13:
We need to investigate that ability to change syncMsgVersion in the register app interface request when a value greater than 3 is received from core.  This would require changing how core determines which RPCs are valid to send to mobile, as well as how mobile communicates which RPCs it supports to the HU.

justinjdickow, 2016-04-12:
@joeljfischer Are you good with closing this?

joeljfischer, 2016-04-12:
Sure

closed 2016-09-22 by jacobkeeler