smartdevicelink/sdl_core #17: [Spec] The version can never be increased beyond 15, or backward compatibility is broken

joeljfischer, 2015-01-23:
We discovered #16 when we were discussing this issue.

We can never actually increase the version beyond 15 because of how backward compatibility works. If we attempt to, we would run into an issue with the Head Unit sending it's max version back as part of a StartSessionACK. If the head unit sends a new style ACK with a larger version byte, then previous versions of the mobile library will not be able to read it. If it sends an old style ACK then that version can never be increased beyond 0x0F.

joeygrover, 2015-01-23:
The other issue here is that v2+ head units send back a v2 packet (12 bytes) not a v1 packet (v1). So if an older app only had the v1 protocol  version of the mobile lib it wouldn't know what to do the extra 4 bytes attached to the ACK and most likely crash. (Possible reason why some older apps were crashing when connecting)

We COULD do something along the lines of if version==16 then the payload actually contains the version in subtag fashion. Since right now there is some data that comes back in the ack (I believe the hash id), but mobile libs can handle it because the data size is correct for that size payload. While I really don't like this solution, I know it would work. Other ideas are much appreciated.

joeljfischer, 2015-01-23:
@joeygrover Your first paragraph should be covered in #16, I think.

joeygrover, 2015-01-23:
You are correct. Didn't see you wrote an issue of it.

justinjdickow, 2016-04-12:
Moving to spec repo

closed 2016-04-12 by justinjdickow