smartdevicelink/sdl_core #32: [Help] SDL Core Internal Error callback func?

yifili09, 2015-02-09:
Hi, Guys.

The response of "RegisterAppInterface" shall indicate whether the app could be list on the Head Unit.

Are there any internal Error callbacks / Notifications from SDL Core to Head Unit such as 
- JSON parsing(Proxy) errors
- App crash erros
- JSON request/response generating erros
- Transiting data corruption
etc

Thanks a lot!

justinjdickow, 2015-02-09:
@yifili09 Could you clarify what you mean by SDL Core to the Head Unit? Also, do you think this issue is a duplicate of #31?

yifili09, 2015-02-10:
Head Unit could be referred to HMI. 
There should be possibility that HMI can not start the application. 

Is there any pre-defined func by SDL Core to meet with the scenario? Thanks.
[image]

justinjdickow, 2015-02-10:
In the current API version no, but in 4.0 we're adding an error and error code to SDL.ActivateApp. When the application is listed but cannot be launched, the error code will return "APPLICATION_NOT_REGISTERED".

The reason this API does not exist in the current version is that an application cannot be listed by SDL unless it is launched on the phone, has been connected, and is already registered with the head unit and at least placed in the HMI_NONE or HMI_LIMITED state. When applications are disconnected or unregistered in the current version they are removed from the list. In 4.0, applications which can be remotely launched and connected will be in the list even if they aren't currently registered.

closed 2015-02-10 by justinjdickow

justinjdickow, 2015-02-11:
@yifili09 Here is more accurate technical information regarding this question

>   1.1. requested app is successfully launched:
        HMI->SDL: SDL.ActivateApp_request (app_1) 
        SDL->app_launcher: OnSystemRequest(launch_app)
        app_1->SDL: RegisterAppInterface 
        SDL->HMI: OnAppRegistered (app_1) 
        SDL->HMI: SDL.ActivateApp_response (app_1)
    In order not to make changes to SDL.ActivateApp, it was decided to reuse OnAppRegistered for providing HMI with registration-related details.

   >1.2. requested app was not launched (=not registered during SDL's timeout started after sending OnSystemRequest(launch_app) to mobile):
        HMI->SDL: SDL.ActivateApp_request (app_1) 
        SDL->app_launcher: OnSystemRequest(launch_app)
        app_1 does not register during timeout
        SDL->HMI: SDL.ActivateApp_error (result: APPLICATION_NOT_REGISTERED)