smartdevicelink/sdl_core #35: libusb_open failed: LIBUSB error access

justinjdickow, 2015-02-19:
I have libusb and aoa flags on, when I connect a USB device I get the following error message:

```
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] AOA protocol version 2
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
INFO  [19 Feb 2015 08:47:11,803][TransportManager] USB control transfer completed
ERROR [19 Feb 2015 08:47:12,707][TransportManager] libusb_open failed: LIBUSB_ERROR_ACCESS
```

chenbd, 2015-02-26:
Is this a permission problem related to udev rules?

justinjdickow, 2015-02-26:
@lea255ace Found a fix at least in the environment I was working in:

Set USB Permissions in `/etc/udev/rules.d/90-usbpermission.rules` by adding

`SUBSYSTEM==“usb”,GROUP=“users”,MODE=“0666”`

closed 2015-02-26 by justinjdickow

pankaj2590, 2016-10-26:
Hi,

I am facing same issue. Above mentioned solution also not working for me on linux i386 platform.
Below is the error message i am getting after AOA connection:

TRACE [26 Oct 2016 12:23:53,960][0xac27db40][TransportManager] /home/pankaj/Projects/smartdevicelink/sdl_core/src/components/transport_manager/src/usb/libusb/usb_handler.cc:110 transport_adapter::UsbHandler::DeviceArrived: enter. libusb_device* 0xb5000f88
ERROR [26 Oct 2016 12:23:53,960][0xac27db40][TransportManager] /home/pankaj/Projects/smartdevicelink/sdl_core/src/components/transport_manager/src/usb/libusb/usb_handler.cc:126 transport_adapter::UsbHandler::DeviceArrived: libusb_open failed: LIBUSB_ERROR_ACCESS
TRACE [26 Oct 2016 12:23:53,960][0xac27db40][TransportManager] /home/pankaj/Projects/smartdevicelink/sdl_core/src/components/transport_manager/src/usb/libusb/usb_handler.cc:127 transport_adapter::UsbHandler::DeviceArrived: exit. Condition: libusb_ret != LIBUSB_SUCCESS
TRACE [26 Oct 2016 12:23:53,960][0xac27db40][TransportManager] /home/pankaj/Projects/smartdevicelink/sdl_core/src/components/transport_manager/src/usb/libusb/usb_handler.cc:243 transport_adapter::ArrivedCallback: exit with 0

I have added above mentioned udev rule and reloaded the rules, but no luck.
Thanks,

ckevar, 2018-06-02:
HI, 
@justinjdickow solution worked for me, Gubuntu 16.04. just consider to re-login.