smartdevicelink/sdl_core #105: Regarding onSystemRequest Notification

ojmakke, 2015-05-05:
Looking at MOBILE_API.XML, 

In function onSystemRequest Notification, there is a URL parameter. The description says:
"If blank, the binary data shall be forwarded to the app.".

Where does the binary data go? I see no parameter in that function to hold binary data.

justinjdickow, 2015-05-05:
Binary Data is part of the [hybrid service](https://github.com/smartdevicelink/protocol_spec#42-hybrid-bulk-data-service)

You can set it in an RPC class using

```
(*message_)[strings::params][strings::binary_data] = binary_data;
```

The [application manager](https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/application_manager_impl.cc#L1801-L1812) reads the binary data off the message and sets it

```
  if (message.getElement(jhs::S_PARAMS).keyExists(strings::binary_data)) {
    application_manager::BinaryData* binaryData =
      new application_manager::BinaryData(
      message.getElement(jhs::S_PARAMS).getElement(strings::binary_data)
      .asBinary());

    if (NULL == binaryData) {
      LOG4CXX_ERROR(logger_, "Null pointer");
      return false;
    }
    output.set_binary_data(binaryData);
  }
```

ojmakke, 2015-05-05:
Thank you. The next question is, at what point will sdl_core identify the RPC as hybrid service? Is that done somewhere in the protocol layer by checking if binary_data is available?

justinjdickow, 2015-05-06:
The optional parameter defaults are in the method signature for the [RawMessage constructor](https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/include/protocol/raw_message.h#L57-L60)

```
  RawMessage(uint32_t connection_key, uint32_t protocol_version,
             const uint8_t *const data_param, uint32_t data_size,
             uint8_t type = ServiceType::kRpc,
             uint32_t payload_size = 0);
```

ojmakke, 2015-05-06:
Thanks for reply again. Yeah i saw that earlier. I believe I didn't explain well what I meant. I was looking for another constructor which defaults to ServiceType::kBulk or some construction which passes ServiceType::kBulk. 

It is not clear where ServiceType is set to kBulk at this point.

I will continue  looking tomorrow with fresh eyes.

justinjdickow, 2015-05-06:
Are you referring to messages that are sent from the phone to sdl core or from sdl core to the phone?

ojmakke, 2015-05-06:
From sdl core to phone. At what point will Service Type be set to 0x0F Bulk Data

justinjdickow, 2015-05-12:
We've confirmed that a message sent from core->mobile is always the RPC service by sending some bulk data in an onSystemRequest after register app interface to the proxy. Will investigate if there are implications of setting the service type correctly...

closed 2015-05-12 by justinjdickow