smartdevicelink/sdl_core #126: Documentation for changing SDL server endpoint.

ssmereka, 2015-05-28:
In order to connect core to an SDL server you must update the default policy JSON file or current mysql policy table database to point to the desired SDL server.  Documentation to do this does not currently exist on github, but should!

ssmereka, 2015-05-29:
This issue is currently blocked by issue #127.

justinjdickow, 2015-06-16:
Added a [policy wiki page](https://github.com/smartdevicelink/sdl_core/wiki/Policies) which addresses this documentation issue.

For convenience the answer is:

SDL uses the 0x07 endpoint defined in the [preloaded policy table](https://github.com/smartdevicelink/sdl_core/blob/develop/src/appMain/sdl_preloaded_pt.json#L16). This endpoint should be changed so that policy table triggers are directed to your own implementation of the SDL Server.

If SDL has already been run and the database has already been created, it should be removed and re-run (build/src/appMain/storage/policy.sqlite) or the database can be edited directly.

closed 2015-06-16 by justinjdickow