smartdevicelink/sdl_core #133: Extra GENIVI repos for SDL

jeremiah, 2015-06-12:
Hi!

GENIVI has both archives of old SDL repos and mirrors of current SDL repos hosted in its git repository: http://git.projects.genivi.org/

We wonder if it makes sense for us to still be carrying the older repos? Is anyone using them? Does the SDL project have the same history in its repos? 

Feedback welcome!

justinjdickow, 2015-06-19:
I'd like to keep the genivi archives because they contain branches that do not exist in the GitHub repository for features that we previously worked on, such as an OS X port. The archives also do not share a parent commit with the GitHub SDL so we consider them different.

gunnarx, 2015-06-19:
No, seriously, it's a complete mess.  We need to take a look at this and sort it out.

There are three(!) repositories that have apparently no commits at all.

In addition, these two appear to have identical content, at least at first look (?):
smartdevicelink_tester_android.git
and
smartdevicelink_tester_android-archive.git

All of the repositories named "test" something have either 9 month old commits or none at all.  Does the current github project not have any such test code for similar purpose, or are also these old history of things that are different now. If these test projects are not receiving updates we should probably name them archive also.   

So, really we need to understand what content is what, and reduce the number of repositories and clarify their purpose.

Also, for trees that do not share a root node, it might still be possible to merge them without losing any history!   See here:
http://stackoverflow.com/questions/1683531/how-to-import-existing-git-repository-into-another

jeremiah, 2015-06-25:
What about empty repos? Like: 
http://git.projects.genivi.org/?p=smartdevicelink_test_android.git;a=tree
http://git.projects.genivi.org/?p=smartdevicelink_test_ios.git;a=tree
http://git.projects.genivi.org/?p=smartdevicelink_tester_ios.git;a=tree

Those can be removed, no?

genivi, 2015-06-25:
> 4058>.

Yes, I mentioned those in my comment too.  Rather than just a quick fix
I'd appreciate if the content/purpose of each one is clearly analyzed and
documented.

Justin, can we reopen this ticket please?

- Gunnar

justinjdickow, 2015-06-25:
@jeremiah Those can be removed, it's the android, android tester, and core projects that I'd like to keep the archives for reference.

jeremiah, 2015-06-25:
Okay. I'll just remove the empty repos from the web interface for now.

jeremiah, 2015-06-25:
So we're now down to;

smartdevicelink.git
smartdevicelink_android-archive.git
smartdevicelink_android.git
smartdevicelink_archive.git
smartdevicelink_ios.git

Can I change the config of the two "archive" repos? I can make it so that they exist on disk, you can clone and pull and everything, but they don't show up in the gitweb interface.  

Would that be satisfactory?

justinjdickow, 2015-06-25:
Sounds good to me

jeremiah, 2015-06-26:
Great. That leaves us with three visible repos and everything is nice and tidy. :-)

I'm fine with closing this bug if everyone else wants to.

closed 2016-06-14 by jeremiah