smartdevicelink/sdl_core #138: Release Branch Fatal Error

JackLivio, 2015-06-17:
I am trying to start the version of SDL Core on the release/4.0.0 branch and am getting a fatal error "HMI index.hml doesn't exist!" 

Is there any setup/configuration that is different from the version of SDL core on the develop branch?

AGaliuzov, 2015-06-18:
There is another opensource project named WebHMI. 
Please find Readme here
https://github.com/LuxoftSDL/sdl_hmi

In short, you have to run HMI(index.html) to make SDL Core workable.

justinjdickow, 2015-06-18:
If this is the new way of separating HMIs, can you please move the sdl_hmi repo into the smartdevicelink GitHub so that everything is in one place and update the sdl_core readme with new instructions...

JackLivio, 2015-06-22:
HMI repo has been moved/merged to smart device link github. S'all goodman.

closed 2015-06-22 by JackLivio