smartdevicelink/sdl_core #148: SDL_Core(Use Qt HMI) Build Error

gmmyok, 2015-06-30:
Hi!
When I compiled SDL-Core(Use Qt HMI) code, I got a build error:

[ 96%] Built target copy_library_HWButtons
make[2]: *** [src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons] Error 127
make[1]: *** [src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/all] Error 2
make: *** [all] Error 2

How can I solve it?
3Q~

jeremiah, 2015-06-30:
Which version of Qt? Which platform are you building for? Which version of SDL?

gmmyok, 2015-07-01:
Qt version: Qt 5
platform: Ubuntu 14.04 LTS
SDL version: 4.0.0-RC2

gmmyok, 2015-07-01:
Has my environment problem, or is CMakeFile with error?

build error detail
///////////////////////////////////////////////////////////////////////////
This program built for x86_64-pc-linux-gnu
Reading makefiles...
Reading makefile `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'...
Reading makefile `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make' (search path) (no ~ expansion)...
Updating makefiles....
 Considering target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make'.
  Looking for an implicit rule for `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make'.
  Trying pattern rule with stem `progress.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make,v'.
  Trying pattern rule with stem `progress.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/RCS/progress.make,v'.
  Trying pattern rule with stem `progress.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/RCS/progress.make'.
  Trying pattern rule with stem `progress.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/s.progress.make'.
  Trying pattern rule with stem `progress.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/SCCS/s.progress.make'.
  No implicit rule found for `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make'.
  Finished prerequisites of target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make'.
 No need to remake target `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/progress.make'.
 Considering target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'.
  Looking for an implicit rule for `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'.
  Trying pattern rule with stem `build.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make,v'.
  Trying pattern rule with stem `build.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/RCS/build.make,v'.
  Trying pattern rule with stem `build.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/RCS/build.make'.
  Trying pattern rule with stem `build.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/s.build.make'.
  Trying pattern rule with stem `build.make'.
  Trying implicit prerequisite `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/SCCS/s.build.make'.
  No implicit rule found for `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'.
  Finished prerequisites of target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'.
 No need to remake target `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build.make'.
Updating goal targets....
Considering target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build'.
 File `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/build' does not exist.
  Considering target file `qmltypes_HWButtons'.
   File `qmltypes_HWButtons' does not exist.
    Considering target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons'.
     File `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons' does not exist.
     Finished prerequisites of target file `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons'.
    Must remake target `src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons'.
Putting child 0x020e0c20 (src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons) PID 5900 on the chain.
Live child 0x020e0c20 (src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons) PID 5900 
Reaping losing child 0x020e0c20 PID 5900 
make[2]: *** [src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons] Error 127
Removing child 0x020e0c20 PID 5900 from chain.
Reaping losing child 0x010fd730 PID 5899 
make[1]: *** [src/components/qt_hmi/qml_plugins/hw_buttons/CMakeFiles/qmltypes_HWButtons.dir/all] Error 2
Removing child 0x010fd730 PID 5899 from chain.
Reaping losing child 0x01e5ad10 PID 5390 
make: *** [all] Error 2
Removing child 0x01e5ad10 PID 5390 from chain.

jeremiah, 2015-07-01:
Looks like you're problem is related to missing files from Qt, not  SDL.

gmmyok, 2015-07-02:
Thank you for your answer.
I will check my QT to find out the problem.

and I have another problem, 
I want to build SDL-Core with QML HMI,
I run cmake "cmake <root_SDL_folder> -DHMI=qt -DHMI2=ON", is it correct?

justinjdickow, 2015-07-02:
Yes, that's correct

justinjdickow, 2015-07-02:
I wrote up some [instructions](https://github.com/smartdevicelink/sdl_core/wiki/SDL-on-Linux-with-QT) to build and run SDL on Linux 14.04 using QT. Don't hesitate to open new issues for any additional questions!

closed 2015-07-02 by justinjdickow

gmmyok, 2015-07-03:
Awesome！
thx a million~