smartdevicelink/sdl_core #157: Can't start communication with SDL core

gmmyok, 2015-07-14:
I want to start communication with SDL core, 
I notify sdl-core that hmi is ready, my code is as follows:

    QDBusMessage msg = QDBusMessage::createSignal("/", "com.ford.hmi.sdl.BasicCommunication", "OnReady");
    QDBusConnection::sessionBus().send(msg);

Is it right? or what need to be done in advance?
But my UI.IsReady is not called.(UI.IsReady has been registered in dbus)

justinjdickow, 2015-07-14:
Can you share your registration code for DBus?

gmmyok, 2015-07-15:
code snippet:
main.cpp

	QDBusConnection::sessionBus().registerService("com.ford.sdl.hmi");
	iautoUI* iautoui = new iautoUI();
	new UIAdaptor(iautoui);
	QDBusConnection::sessionBus().registerObject("/", iautoui);

iautoui.h

	class iautoUI : public QObject
	{
	    Q_OBJECT
	    Q_CLASSINFO("D-Bus Interface", "com.ford.sdl.hmi.UI")
	public:
	    explicit iautoUI(QObject *parent = 0);

	public slots:
	    void IsReady();
	};

iautoui.xml

	<node>
	  <interface name="com.ford.sdl.hmi.UI">
	    <method name="IsReady">
	    </method>
	  </interface>
	</node>

justinjdickow, 2015-07-20:
I'm not sure this matters, but can you try putting `registerService` after `registerObject`?

gmmyok, 2015-08-18:
Resovled it.
I sent QDBusMessage by error param.
The correct path is "com.ford.sdl.hmi.BasicCommunication", but not "com.ford.hmi.sdl.BasicCommunication".

Thx for answer.

closed 2015-08-18 by gmmyok