smartdevicelink/sdl_core #166: Policy Table Update Process (Ford Vs Genivi Impl)

JackLivio, 2015-07-21:
The policy table update process outlined in the [Software Detailed Design] (https://app.box.com/s/ohgrvemtx39f8hfea1ab676xxrzvyx1y) on pages 36 and 37, shows the Genivi specific implementation and the Ford specific implementation.

Currently the SDL Core and SDL Server repositories implement the Ford specific version of updating the policy table, and due to limitations of the Web browser, the update cannot be made without a "workaround".

If you look at the diagrams in the Software detailed design document, the Genivi implementation for a policy table update does not require the HMI to parse/decrypt the policy table update data received from SDL Server. The Genivi implemntation instead uses a put file request which then sends the policy table update to the policy handler via ReceiveMessageFromSDK(file_path, content). 

In the diagram, notice that the function ReceiveMessageFromSDK(file_path, content) is used by both Ford's and Genivi's implementation and the rest of process following that function call is the same for both implementations.

The differences between the two implementations are:
- The Genivi specific implementation for receiving a policy table update is initiated by a put file request.
- The Ford specific implementation for receiving a policy table is initiated by a system request.
- Genivi's design does not specifically mention policy table encryption.
- Ford's design requires the HMI to decrypt the policy table and write the decrypted data to a new file(Main problem since the web browser cannot do this!)
- Genivi's design requires less steps and does not require the HMI to do anything (Makes more sense for development/testing).

Question: Why are there two different implementations for the same process?

I would like to open a discussion for choosing one of these processes, or perhaps redefine a new process as the "standard" for updating a policy table.

My suggestion for a possible implementation: 

- Initiate policy table update via system request from server (Same as Ford's design)
- Using #ifdefs within SystemRequest::Run(), if the policy table encryption #ifdef is enabled, continue with Ford's implementation and pass the file path to the hmi to be 'decrypted'. 
- Alternatively, If policy table encryption #ifdef is disabled, then parse the incoming data within SystemRequest::Run() and then call ReceiveMessagaeFromSDK(file_path, content) immediately (similar to Genivi's design except the put file request has been replaced by the system request rpc).

JackLivio, 2015-07-21:
Also after looking at put_file_request.cc at line [115](https://github.com/smartdevicelink/sdl_core/blob/release/4.0.0/src/components/application_manager/src/commands/mobile/put_file_request.cc#L115), the logic for identifying the put file request as a policy table update seems to be unfinished. Someone left a todo that has yet to be implemented

Code:
```
// Policy table update in json format is currently to be received via PutFile
// TODO(PV): after latest discussion has to be changed
if (mobile_apis::FileType::JSON == file_type_) {
policy::PolicyHandler::instance()->ReceiveMessageFromSDK(sync_file_name_, binary_data);
  }
```

JackLivio, 2015-07-28:
Resolved issue by adding an #ifdef to the PolicyManagerImpl::LoadPT function. If the #ifdef USE_HMI_PTU_DECRYPTION is enabled, the policy table update process  will operate as per Ford's specification. If the #ifdef is disabled, then the policy manager uses a different parse function to get the data from an unencrypted array received from SDL server.

justinjdickow, 2015-08-06:
@JackLivio Was your fix merged into the release branch before this was closed?

closed 2016-01-19 by justinjdickow