smartdevicelink/sdl_core #167: DCHECK in release/4.0.0 crashes core

JackLivio, 2015-07-21:
In message_helper.cc, the DCHECK() calls at lines [1305 and 1307](https://github.com/smartdevicelink/sdl_core/blob/release/4.0.0/src/components/application_manager/src/message_helper.cc#L1305) crash core on an assert. Whether core crashes at line 1305, or crashes on 1307 depends on which tester app I am using.

If I comment out both of those lines and recompile Core, everything works fine. 

Why is DCHECK being used in the release branch?

AGaliuzov, 2015-07-29:
I believe the lines mentioned above should be removed, because in lines [1348 and 1351](https://github.com/smartdevicelink/sdl_core/blob/release/4.0.0/src/components/application_manager/src/message_helper.cc#L1348) we perform double check for these variables.

Actually assert has performed because application didn't sent either ngn_media_screen_name or application types.

justinjdickow, 2015-07-31:
@JackLivio were you using a SPT distribution from hockey app that didn't include ngn_media_screen_name?

JackLivio, 2015-08-06:
I am not sure if it included ngn_media_screen_name, I do not have the source code for the version of SPT that was experiencing the crash. The app I was using was a version  of SPT from Luxoft that has support for video streaming.

On the Open SDL Call a couple of weeks ago someone was having an issue with the app crashing on the DCHECK() two lines before:
```
DCHECK_OR_RETURN(app_types, false);
```
I do not know the details of the app they were using.

Do we want SDL to just crash itself because the connected app does not have the proper information? Can't we just reject the connection?

justinjdickow, 2015-08-13:
@JackLivio The DCHECK only happens in debug builds so it won't crash itself when it matters, but I generally agree anyway, we should return useful error messages.

PR #337 (2016-01-25): Remove DCHECKs that seem to always be tripping everyone up
Fixes #167
closed 2016-01-25 by AGaliuzov