smartdevicelink/sdl_core #187: [Old Version] A latency happened on USB Transport Layer

yifili09, 2015-08-07:
In 3.6, there is an issue report about latency on SDL decoding raw data.
Please review and help to identify whether it was a known issue or not. 
Thanks!

## Quote

>We were doing some timing measurement to see the video latency.
>We have noticed the following behavior, can you please check this issue? is there some known fix?

>the event listener thread in ***Transport_manager_impl.cc*** is not running for approximately 300 ms, we can see the "Data received" event is raised by adapter listner as soon as the data is read from the USB. So the data read from USB is not processed by HU(SDL_CORE) for approximately 300 ms.
```
[115940] 2:8; applink_usb.c:39; WARN - data size is 15988
```
>the event raised by adapter listner (in function ***TransportAdapterListenerImpl::OnDataReceiveDone***)
USB transfer DONE (in ***USB connection.cc***)
```
[116267] 2:8; applink_usb.c:39; WARN - data size is 15976
```

>the event raised by adapter listner (in function ***TransportAdapterListenerImpl::OnDataReceiveDone***)
USB transfer DONE (in ***USB connection.cc***)
```
[116267] 2:8; applink_usb.c:39; WARN - data size is 0
```

>Event listener thread run (in ***Transport_manager_impl.cc,TransportManagerImpl::EventListenerThread***)
the packet received from TM in Protocol Handler (***ProtocolHandlerImpl::OnTMMessageReceived***)
message raised as event in TM to Protocol Handler (***TransportAdapterListenerImpl::EventTypeEnum::ON_RECEIVED_DONE:***)
Event listener thread run (in ***Transport_manager_impl.cc , TransportManagerImpl::EventListenerThread***)
the packet received from TM in Protocol Handler (***ProtocolHandlerImpl::OnTMMessageReceived***)
message raised as event in TM to Protocol Handler (***TransportAdapterListenerImpl::EventTypeEnum::ON_RECEIVED_DONE:***)

henry-fan, 2015-08-27:
@justinjdickow 
Just wondering if there's anyone has had a chance to look into the issue?

justinjdickow, 2015-09-03:
@henry-fan I'm in the process of implementing video streaming on iOS, and the Android video streaming that we've tested (now merged into develop), does not seem to exhibit the behavior. If I see an issue in iOS streaming latency I will report back here.

LuxoftAKutsan, 2017-12-01:
Priority low:
Looks like it is not SDL issue, and can be reproduces only via Ios

JackLivio, 2018-02-26:
@yifili09 Closing due to using version 3.6 which is not available on the open source repository. If this issue is still being experienced on more recent versions of core please let me know and i will reopen this issue.

closed 2018-02-26 by JackLivio