smartdevicelink/sdl_core #195: How to choose between Ford's Applink and SmartDeviceLink, what's the difference?

kingtheus, 2015-08-19:
1. If we are smartphone app developers ， which library should we use， Applink libs or sdl libs？
    what's the difference between them? 
   my understanding is :
   (1)applink is private so we can only use what it has provided， but with sdl libs，you can implement      your requirements if it hasn't provided.  
   (2) applink might show up Ford's Logo in your app which sdl might not?
   (3) app publish process might be different? if we use AppLink, our app must be reviewed by Ford otherwise we might publish our app freely?

if i were wrong please correct me , and any other differences?

Sincerely Jin

jackwilliambell, 2016-02-19:
Has this issue been resolved? I'm curious about this things as well.

JackLivio, 2018-03-01:
I think this question would be better addressed by an issue in the IOS or Android SDL repositories. 

To answer your questions
(1) If you make changes to the libraries there is no garuntee it will work on different head units. Please consider submitting an evolution proposal if you would like to say mobile library changes be supported by the sdl eco system. https://github.com/smartdevicelink/sdl_evolution
(2) The logo during a lock screen is dependent on the connected head unit. There is also no an ecosystem of sdl consortium members that support the project and this includes several OEMs other than Ford.
(3) Application verification process has not been fully completed by the SDLC and I believe each OEM such as Ford have their own verification process since each OEM controls enforces their own app policy requirements via the sdl policy server and policy table.

Please refer further questions to the sdl slack: http://slack.smartdevicelink.com

closed 2018-03-01 by JackLivio