smartdevicelink/sdl_core #196: How to transfer multi screens?

gmmyok, 2015-08-21:
I want to develop multi screens in the same App HMI.
(such as A screen and B screen)

First, A screen is displayed, 
user presses a button of A screen, 
and then B screen is displayed
(the screen is transferred from A to B)

Dose the SmartDeviceLink framework support it?
How to do it?

gmmyok, 2015-08-24:
@justinjdickow Could you help me or give me some advice?

stefanek22, 2015-08-24:
Although this is ultimately dependent upon a platform's HMI support, the overall concept is supported by any platform that integrates the SetDisplayLayout RPC.  Using that RPC, you can switch to other defined templates.

gmmyok, 2015-08-25:
Basically I know what you mean, and I will investigate the SetDisplayLayout RPC.
thx a lot.

closed 2016-01-19 by justinjdickow