smartdevicelink/sdl_core #199: Wrong use of 'device_info' for UpdateDeviceList and OnDeviceChosen

JackLivio, 2015-08-24:
## Issue
When selecting a device from the HMI's "Change Devices" option, I get the error from core "Incorrect parameter from HMI". After doing some investigating I found that OnDeviceChosen is sending back the mac address of the selected device in a string to core. It is being rejected by core because the expected parameter should be an integer. Also this parameter shouldn't be the mac address of the device, but the connection handler id instead.

## Current RPC Behavior
UpdateDeviceList sends an updated list of connected devices to the HMI and stores their "DeviceInfo" for later use. When a device is selected from that device list on the HMI, the notification OnDeviceChosen is sent to core indicating that the user wants to connect SDL to that device. Both of these RPC's are sending the device's mac address as the 'id' parameter.

Core then attempts to find the device and connect to it by using the connection handle id, not the mac address.

```
void OnDeviceChosenNotification::Run() {
  LOG4CXX_AUTO_TRACE(logger_);

  if ((*message_)[strings::msg_params].keyExists(strings::device_info)) {
    ApplicationManagerImpl::instance()->ConnectToDevice(
        (*message_)[strings::msg_params][strings::device_info][strings::id]
            .asInt());
  }
}

```
```
void ApplicationManagerImpl::ConnectToDevice(uint32_t id) {
  // TODO(VS): Call function from ConnectionHandler
  if (!connection_handler_) {
    LOG4CXX_WARN(logger_, "Connection handler is not set.");
    return;
  }

  connection_handler_->ConnectToDevice(id);
}
```
```
void ConnectionHandlerImpl::ConnectToDevice(
    connection_handler::DeviceHandle device_handle) {
  connection_handler::DeviceMap::const_iterator it_in;
  it_in = device_list_.find(device_handle);
  if (device_list_.end() != it_in) {
    LOG4CXX_INFO_EXT(logger_,
                     "Connecting to device with handle " << device_handle);
    if (transport_manager_) {
      if (transport_manager::E_SUCCESS
          != transport_manager_->ConnectDevice(device_handle)) {
        LOG4CXX_WARN(logger_, "Can't connect to device");
      }
    } else {
      LOG4CXX_ERROR(logger_, "Null pointer to TransportManager.");
    }
  } else {
    LOG4CXX_ERROR(
        logger_, "Application Manager wanted to connect to non-existing device");
  }
}

```

### Fix
Change the RPC message formation of UpdateDeviceList to use the connection handle id instead of mac address. 

### Concern/Question
For security reasons, it is important for the hashed mac address to be used here? If so the connection logic invoked by OnDeviceChosen must be changed to support it.

AGaliuzov, 2015-08-28:
@justinjdickow ,
Please take a look on PR #179 it is about to fix the described beahviour. Please let me know in case of any questions

JackLivio, 2015-08-28:
Yes this looks good. We would like to have the device_info as a string added to the release/4.0.0 branch as a hotfix. I will open a pull request for this shortly. Just a heads up if this creates a merge conflict with develop down the road.

closed 2016-01-19 by justinjdickow