smartdevicelink/sdl_core #263: [SDL3_5] in SDL 3.5 version reset occasionally

jinujose50, 2015-11-16:
Seems like the below code in bool AddCommandRequest::CheckCommandName(ApplicationConstSharedPtr app) function cause reset of radio sometimes (May be once in 300 times).

The SDL code version is 3.5.

  for (; commands.end() != i; ++i) {

    if(!(*i->second).keyExists(strings::menu_params)) {

      continue;

    }

The code is modified in latest version, is this reset fixed in latest version??

The issue is observed when APP is getting registered. What could be the possible root cause for this reset?

justinjdickow, 2015-11-17:
Do you have a backtrace for when the crash happens?

Based on the code above the only thing that comes to mind is that the commands map is being accessed and modified elsewhere. While removing an element would not invalidate the iterator every time, it would invalidate the iterator in the case that it is pointing to the member that was erased, which could explain the very low occurrence rate.

jinujose50, 2015-11-17:
yes, please find the backtrace
FAULT - Stack backtrace: 0xe640 0x2c52da40 0x2c277d30 0x2b03b304 0x2b039ff4 0x2af856b8 0x2af89a38 0x2af8b31c 0x2afb1104 0x2afb074c 0x2c15f724

2c500000-2c619000 r-xp 00000000 b3:01 552        /lib/libc-2.13.so
NsSmartDeviceLink::NsSmartObjects::SmartObject::keyExists(std::string const&) const
smart_object.cc:973

application_manager::ApplicationManagerImpl::ManageMobileCommand(utils::SharedPtr<NsSmartDeviceLink::NsSmartObjects::SmartObject> const&)
application_manager_impl.cc:1360
application_manager::ApplicationManagerImpl::ProcessMessageFromMobile(utils::SharedPtr<application_manager::Message> const&)
application_manager_impl.cc:1730
application_manager::ApplicationManagerImpl::Handle(application_manager::impl::MessageFromMobile const&)
application_manager_impl.cc:1945
threads::MessageLoopThread<utils::PrioritizedQueue<application_manager::impl::MessageFromMobile> >::LoopThreadDelegate::DrainQue()
message_loop_thread.h:161
threads::MessageLoopThread<utils::PrioritizedQueue<application_manager::impl::MessageFromMobile> >::LoopThreadDelegate::threadMain()
message_loop_thread.h:145
threadFunc
posix_thread.cc:50

closed 2016-10-28 by JackLivio