smartdevicelink/sdl_core #279: does SDL4.0 support protocol version 1.0 ?

kon-shine, 2015-11-30:
I checked application_manager_impl.cc  
the code inside  "case ProtocolVersion::kV1:"(line:1660)  would always return false. 
 
Could  SDL4.0 be backward compatible with protocol version 1.0 ?

Thanks

justinjdickow, 2015-12-14:
@kon-shine Can you explain how that code always returns false?

kon-shine, 2015-12-17:
There is an error log :"Message received: UNSUPPORTED_VERSION" when a V1 message coming, and if this's a well-formed V1 message, false would returned.
please see the check steps for details: 
1, I compared HandleIncomingMessageProtocolV1() and  HandleIncomingMessageProtocolV2() in moblie_message_handler.cc:84 -179 and found that no set_function_id() is called in HandleIncomingMessageProtocolV1(). so here function_id_ of Message for V1 is its initial value 0.(its constructor initializer list: message.cc,line 59 )
2, so in line 1666 of application_manager_impl.cc, message.function_id() is 0 .
when a V1 message coming, there is alway a LOG4CXX_ERROR about "Message received: UNSUPPORTED_VERSION"(line:1667) 
3. if conversation_result is kSuccess,false is returned.(application_manager_impl.cc:1675,1693)

justinjdickow, 2016-01-19:
I've confirmed that SDL does not support protocol version 1.

closed 2016-01-19 by justinjdickow