smartdevicelink/sdl_core #307: after running 'cmake' command getting following error

anupamwcities, 2015-12-24:
Hi ,
I am installing sdlcore on my linux OS , getting following error while running 'cmake' command.
any help will be appreciated .

error message -

 [ 12%] Building CXX object src/components/transport_manager/CMakeFiles/TransportManager.dir/src/tcp/tcp_client_listener.cc.o
In file included from /sdl_core/src/components/transport_manager/src/tcp/tcp_client_listener.cc:45:
/usr/include/linux/tcp.h:72: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:72: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:73: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:73: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:74: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:74: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:75: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:75: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:76: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:76: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:77: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:77: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:78: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:78: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:79: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:79: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:80: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:80: error: a function call cannot appear in a constant-expression
/usr/include/linux/tcp.h:81: error: ‘__u32 __fswab32(__u32)’ cannot appear in a constant-expression
/usr/include/linux/tcp.h:81: error: a function call cannot appear in a constant-expression
make[2]: *** [src/components/transport_manager/CMakeFiles/TransportManager.dir/src/tcp/tcp_client_listener.cc.o] Error 1
make[1]: *** [src/components/transport_manager/CMakeFiles/TransportManager.dir/all] Error 2

justinjdickow, 2016-01-19:
What version of GCC are you using?

BrandonHe, 2016-02-24:
Maybe you can try include netinet/tcp.h instead linux/types or do  s/__cpu_to_be32/__constant_cpu_to_be32/ in <linux/tcp.h> to avoid this error. Hope it's useful.

jacobkeeler, 2016-10-27:
Closing due to no response

closed 2016-10-27 by jacobkeeler