smartdevicelink/sdl_core #357: [SDL4.0 - Question] How to build a SDL app which supports video streaming?

Manabu-GT, 2016-01-27:
Hi,

I'm trying to test the video streaming feature via wifi connection based on https://github.com/smartdevicelink/sdl_core/wiki/Video-Streaming-Setup tutorial.

However, the MobileNavi app mentioned in the document is not available and it's not clear how to build such an app and test with SDL core.
Could you please provide a sample code snippet or app to test the video streaming functionality?

Manabu-GT, 2016-02-01:
Figured it out by looking at the SDL Mobile library.

closed 2016-02-01 by Manabu-GT

jay-manikandan, 2016-12-03:
Hi Manabu,

Can u please share the details about Mobile Navi? I am looking at the sdl-android library and it is not present.

Thanks.

moguli, 2017-01-17:
Hi @Manabu-GT 

I have the same problem in this wiki.
https://github.com/smartdevicelink/sdl_core/wiki/Video-Streaming-Setup
Could you share MobileNavi source code and APK?

Manabu-GT, 2017-01-19:
Hi @moguli and @jay-manikandan,

Sorry, I no longer have access to the source/APK because I already left the company where I worked on SDL. Ideally, it should come from Livio. Maybe ask someone at Livio to add video streaming feature in their https://github.com/smartdevicelink/sdl_tester_android app.