smartdevicelink/sdl_core #364: How to build sdl for embedded linux environemt

tulip0425, 2016-01-29:
we have set up yocto environment on an i.MX6 evaluation board, and would like to run sdl on that board. Is there any guide or demo showing how to modify the CMake thus to build the sdl on embedded linux board? 
Thanks.

JackLivio, 2016-01-29:
We are working on getting sdl setup to run in a yocto environment. Once we have it running I will let you know about what changes were made in the cmake file. 

What architecture type does your eval board use?

tulip0425, 2016-02-02:
Great thanks.
The main cpu is i.MX6dl with dual ARM Cortex A9 at 1G/1.2G Freq.
We are also working on migrating the sdl to this embedded yocto environment but met many problems.
Looking forward to your good news.

jacobkeeler, 2017-04-18:
PR for AGL Port: #1159

closed 2017-04-25 by jacobkeeler