smartdevicelink/sdl_core #370: Show and Putfile Behavior

madhuy, 2016-02-02:
As per spec (6.19.2.2 OnPutFile. File is uploaded after RPC is used) , Show should succeed even image file is updated.

But SDL core implementation returns INVALID_DATA if no file is uploaded prior to show
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/show_request.cc#L98

Please let us know behavior and update spec or SDL core

justinjdickow, 2016-03-21:
The current behavior in master is correct, and will be changed in 4.1 according to HMI Guidelines. We'll have better versioning for the SDL HMI Guidelines with the launch of the developer portal so you'll be able to understand specific versions for any particular behavior. We'll locate the current behavior and upload here

madhuy, 2016-07-13:
@justinjdickow  Can we close this? or keep open by labelling as "Enhancement"

justinjdickow, 2016-07-13:
Since this is fixed in release 4.1.0, I believe we can close it. Documentation is just fragmented and reflects 4.1.0 behavior instead of 4.0.x

closed 2016-07-13 by justinjdickow