smartdevicelink/sdl_core #377: Make step fails on logger, even though liblog4cxx* not installed

jackwilliambell, 2016-02-05:
Having a problem building sdl_core. 

I believe I have properly installed all dependencies and have removed liblog4cxx*. (More accurately, I did a 'sudo apt-get remove liblog4cxx*' and it reported there was nothing to remove.) 

So far as I can tell I have followed all the steps in the readme properly. I have searched the Internet and the mailing list looking for anyone else reporting the same errors, but no joy.

System: 
* Intel® Core™ i5-3210M CPU @ 2.50GHz × 4  
* 3.7 GiB
* Ubuntu 14.04 LTS (64 bit)

cmake output:
```
-- USB support is enabled
-- Bluetooth support is enabled
-- Avahi support is enabled
USE DHMI_PTU_PARSER
-- Found Sqlite3: /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jbell/sdl/sdl_core
```

Error lines from make:
```
collect2: error: ld returned 1 exit status
make[9]: *** [trivial] Error 1
make[8]: *** [all-recursive] Error 1
make[7]: *** [all-recursive] Error 1
make[6]: *** [all-recursive] Error 1
make[5]: *** [src/3rd_party/apache-log4cxx-cmake/CMakeFiles/liblog4cxx] Error 2
make[4]: *** [src/3rd_party/apache-log4cxx-cmake/CMakeFiles/liblog4cxx.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [CMakeFiles/3rd_party_logger] Error 2
make[1]: *** [CMakeFiles/3rd_party_logger.dir/all] Error 2
make: *** [all] Error 
```

Full make output attached.
[makeoutput.txt](https://github.com/smartdevicelink/sdl_core/files/119979/makeoutput.txt)

jackwilliambell, 2016-02-05:
Note: I get the same results when exporting CXX as either 'gcc' or 'g++'.

justinjdickow, 2016-02-06:
Do you have all these dependencies? `cmake build-essential libavahi-client-dev libsqlite3-dev chromium-browser libssl-dev libudev-dev libgtest-dev libbluetooth3 libbluetooth-dev bluez-tools gstreamer1.0* libpulse-dev`

jackwilliambell, 2016-02-06:
As it turns out I did not have libssl-dev, libudev-dev, libgtest-dev, libbluetooth3, libbluetooth-dev, bluez-tools, gstreamer1.0*, or libpulse-dev. Should I enter a second ticket to get those dependencies added to the readme and/or cmake tests?

However, installing those did not fix the problem. Make output attached, but its basically the same as before. 

[makeoutput.txt](https://github.com/smartdevicelink/sdl_core/files/120419/makeoutput.txt)

jackwilliambell, 2016-02-06:
Interesting, I did a 'make clean' and then a 'make' using 'g++' as the CXX export. I am now getting the following make output:
```
[  1%] Built target jsoncpp
 Need to build logger. 
[100%] Built target expat
[100%] Generating ../apr-build/include/private/apr_escape_test_char.h
Error copying file "include/private/apr_escape_test_char.h" to "/home/jbell/sdl/sdl_core/src/3rd_party/apr-cmake/../apr-1.5.0/include/private".
make[5]: *** [src/3rd_party/apr-cmake/../apr-build/include/private/apr_escape_test_char.h] Error 1
make[4]: *** [src/3rd_party/apr-cmake/CMakeFiles/libapr-1.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [CMakeFiles/3rd_party_logger] Error 2
make[1]: *** [CMakeFiles/3rd_party_logger.dir/all] Error 2
make: *** [all] Error 2
```

jackwilliambell, 2016-02-06:
Note: Using 'gcc' as the CXX export produces the same output.

jackwilliambell, 2016-02-11:
I was finally able to get back to this today and made some progress...

The fix for the last problem I had was to start over from a fresh git clone of sdl_core, but with all the dependencies described by Justin above. I ran cmake, make, and make install without problems.

NOTE: I exported CXX as 'g++' for this build.

I now have a new problem: I can't get it to run. Here is the result:
```
$ ~/sdl/sdl_core/src/appMain$ ./smartDeviceLinkCore
./smartDeviceLinkCore: error while loading shared libraries: liblog4cxx.so.10: cannot open shared object file: No such file or directory
```

Has anyone seen this before? Any suggestions?

In any case, it seems to me you can close this issue as 'work-around available'. The work-around is to make sure you have ALL the dependencies and then start fresh.

jackwilliambell, 2016-02-11:
Turns out the fix was: 'apt-get install liblog4cxx*'. Which is exactly the opposite of what the readme recommends. I now have a working sdl_core installation.

Justin, et al: you guys can close this issue. But I do request you update the readme to reflect the extra dependencies and the fact you may need to install liblog4cxx in some cases.

sandeepf184, 2016-09-06:
I have similar problem building sdl_core.
I followed all the steps mentioned in the instructions and also I got all the dependencies. Also tried installing and uninstalling liblog4cxx. Still i'm getting below error. 

syerlank@syerlank-VirtualBox:~/Sandeep/sdl/sdl_core_build/src/appMain$ ./smartDeviceLinkCore 
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "SmartDeviceLinkCoreLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "TransportManagerLogFile".
log4cxx: No appender could be found for logger (SDLMain).
log4cxx: Please initialize the log4cxx system properly.

closed 2016-10-28 by JackLivio