smartdevicelink/sdl_core #401: PolicyHandler::SendMessageToSDK using unimplemented container

JackLivio, 2016-03-01:
In the Master Branch, [PolicyHandler::SendMessageToSDK](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/policies/policy_handler.cc#L728-L763) is using an unimplemented container 'last_used_app_ids_'. 

SendMessageToSDK() will always return false since last_used_app_ids_ is always empty. 

[The develop branch uses GetAppIdForSending() instead of 'last_used_app_ids_'](https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/policies/policy_handler.cc#L749). I think master hasn't been updated with these changes yet, but this is a blocking issue for some developers at the moment and needs to be updated asap.

madhuy, 2016-03-01:
need this change as well. other url been not sent to Mobile https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/message_helper/message_helper.cc#L2093

closed 2016-03-22 by JackLivio

FStoikov, 2019-02-18:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-2867