smartdevicelink/sdl_core #402: Develop Branch Policy Update

JackLivio, 2016-03-02:
I am seeing some differences for the policy table update process in the Develop and Master Branch.

[Commit to develop branch in question] (https://github.com/smartdevicelink/sdl_core/commit/5abb7cdca5e23e9765b616fef5440fe4b0e49d31)

Changes to ../mobile/on_system_request_notification.cc: A part of code was removed where core reads the file contents of the filename passed as a parameter from the hmi. This file that used to be read is supposed to have the entire HTTPRequest header with the policy table parsed as binary data in the request body. In addition, when an onSystemRequest is initiated by the hmi, the removed code was the only place where any bulk data (binary data) would be added to the message parameters to be sent to mobile. 

I see the comments included in this commit that say
```
 +    /* According to requirements:
 +       "If the requestType = PROPRIETARY, add to mobile API fileType = JSON
 +        If the requestType = HTTP, add to mobile API fileType = BINARY"
 +       Also in Genivi SDL we don't save the PT to file - we put it directly in binary_data */
```

Highlighting the last comment, is this repository only supposed to follow Genivi Requirements? What about supporting existing Ford / legacy processes?

JackLivio, 2016-03-02:
Here is a diagram of the ford process that this commit breaks.
[image]

madhuy, 2016-03-02:
with develop branch PT , even it is not working non-ford version PT update . because mobile proxies expecting FORD specific format in JSON (ex: HTTPRequest, headers,body). 
https://github.com/smartdevicelink/sdl_ios/blob/develop/SmartDeviceLink-iOS/SmartDeviceLink/SDLProxy.m#L361

https://github.com/smartdevicelink/sdl_ios/blob/develop/SmartDeviceLink-iOS/SmartDeviceLink/SDLProxy.m#L476

JackLivio, 2016-03-03:
@madhuy: We are working on adding HTTP SystemRequest (Genivi) support to the ios proxy today. We plan to have the mobile libraries construct the HTTP Request header and body for http onSystemRequests.

madhuy, 2016-03-03:
Thanks, I modified SDLproxy locally it working for me . 

And also I noticed , SDL core master branch does not work ford specific implementation also. 
Ford specific PT exchange start from PTExchangeHandlerExt::Start() (pt_exchange_handler_ext.cc). this file not been added in make for compiling.  

Please check this as well.

theresalech, 2016-10-11:
Fixed with [Release 4.0.12](https://github.com/smartdevicelink/sdl_core/releases/tag/4.0.12)

closed 2016-10-11 by theresalech