smartdevicelink/sdl_core #403: How to connect to other Apps besides Livio SDL tester

tulip0425, 2016-03-03:
Hi,
Besides Livio SDL tester, How can i test connecting with other Apps which support Applink?
For example,  those apps officially announced supporting Applink on Ford Website, like spotify, glympse, or others.
thanks.

JackLivio, 2016-03-16:
What transport method are you using to connect to sdl core? If you are using android, most production apps require you to connect via bluetooth/usb. If you are connecting via a tcp connection then apps like Spotify will not connect to core.

tulip0425, 2016-03-22:
Yes, I was using Android and WiFi/TCP connection to connect to core.
I have read through the mailing list that USB connection was not supported yet, right?
So does sdl core support connecting production apps via bluetooth?

thanks.

jacobkeeler, 2017-04-18:
Yes, all production Android apps use bluetooth as their transport.

closed 2017-04-18 by jacobkeeler