smartdevicelink/sdl_core #412: Dialnumber API support

madhuy, 2016-03-11:
This Dial number API is not supported in master but it does support in develop branch. 
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/dial_number_request.cc#L52

justinjdickow, 2016-03-21:
Since this is something that was included in the API in 4.0 but not fully implemented, we can consider its implementation a hot fix for 4.0.x. We're looking into getting this work scheduled. The dial_number_request.cc implementation is obviously not the only change that needs to be made. We have to support the associated HMI API as well as the rpcs in the command factories.

madhuy, 2016-06-15:
This is been fixed. @justin Can we close this ?

closed 2016-06-15 by madhuy