smartdevicelink/sdl_core #416: Slider timeout error code showing as "wrong language" when it should be "time out"

jgrix, 2016-03-11:
-Bug Report
let slider timeout, result code should be "time out", SPT show result code as "wrong language".

-Reproduction Steps
1. Pair phone with web HMI, run SPT3, create a media app proxy.
2. Send slider message RPC with a time out
3. wait for slider time out

-Expected Behavior
response result code should be "time out"

-Observed Behavior
response result code is "wrong language" in SPT. HMI is sending code 10 which according to page 241 of the integration guide lines is the correct error code. 

-OS & Version Information
SDL_Core version: 4.0.4
Phone Type/Model and OS version: Sony Xperia Z3v(Model D6708) with Android version 4.4.4
Mobile Application name and version: SyncProxyTester
connection type: Wifi/Bluetooth/USB: USB
Virtual Box Host Machine OS: Windows 7 professional with Virtual Box running Ubuntu 14.04

JackLivio, 2016-03-11:
The result code "10" on the HMI_API.xml side corresponds to "TIMED_OUT". On the MOBILE_API.xml side the 10th enum for result code is "WRONG_LANGUAGE". I think when core gets the result code from the hmi, it is then indexing the mobile api result enums with the response code when constructing the response to mobile.

madhuy, 2016-03-11:
I see similar behavior in some other API's like 
-ChangeRegistration
-getVehicleData
-setGlobalProperties
-performInteraction
-ScrollableMessage
-Slider

Reason is , hmi_response::code is directly assigned to mobile_apis result, instated of converting by using GetMobileResultCode() as below

mobile_apis::Result::eType result_code =
          GetMobileResultCode(static_cast<hmi_apis::Common_Result::eType>
          (message[strings::params][hmi_response::code].asInt()));

JackLivio, 2016-03-11:
In [slider_request.cc](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/slider_request.cc#L132-L148)

...after receiving the timeout response from the hmi, core gets the repsonse_code as an integer from the message parameters

```
  const int response_code =
      message[strings::params][hmi_response::code].asInt();
```

and later uses this integer to look up the response code corresponding to the mobile_api spec

```
  SendResponse(is_response_success,
               mobile_apis::Result::eType(response_code),
               0,
               &response_msg_params);
```

Since the result codes in the mobile and hmi api's are in different orders and because the same integer value is used to look up the result code, we are seeing the incorrect result reported to mobile. 

### Hotfix on the way!

JackLivio, 2016-03-21:
Hotfix was merged into master

closed 2016-03-21 by JackLivio