smartdevicelink/sdl_core #417: click cancel button on slider, result code should be “aborted”, SPT show result code as invalid id

zyang46, 2016-03-11:
-Bug Report
click cancel button on slider, result code should be “aborted”, SPT show result code as invalid id.

-Reproduction Steps
1. Pair phone with emulator via Bluetooth, run SPT3, create a Bluetooth media app proxy.
2. Send slider message RPC
3. click cancel button or back button

-Expected Behavior
response result code should be "Aborted"

-Observed Behavior
resposne result code is invalid_id

-OS & Version Information
Phone Type/Model and OS version: Sony Xperia Z3v(Model D6708) with Android version 4.4.4
Mobile Application name and version: SyncProxyTest3
connection type: Wifi/Bluetooth/USB: Bluetooth
Virtual Box Host Machine OS: Windows 7 professional with Virtual Box running Ubuntu 14.04
Testing Against/ test case: slider

JackLivio, 2016-03-11:
In [slider_request.cc](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/slider_request.cc#L132-L148)

...after receiving the timeout response from the hmi, core gets the repsonse_code as an integer from the message parameters

```
  const int response_code =
      message[strings::params][hmi_response::code].asInt();
```

and later uses this integer to look up the response code corresponding to the mobile_api spec

```
  SendResponse(is_response_success,
               mobile_apis::Result::eType(response_code),
               0,
               &response_msg_params);
```

Since the result codes in the mobile and hmi api's are in different orders and because the same integer value is used to look up the result code, we are seeing the incorrect result reported to mobile. 

### Hotfix on the way!

JackLivio, 2016-03-21:
Hotfix Merged

closed 2016-03-21 by JackLivio