smartdevicelink/sdl_core #418: SDL is core is allowing 'TextField' text  more than  Max value(500)

madhuy, 2016-03-11:
The Max value  is 500 for textfield . but SDL core is sending to HMI even mobile app sends more than 500.
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L1614

I am not sure who should control it . is it HMI or is it SDL core? 

Please clarify.

JackLivio, 2016-03-11:
I think that value is for 500 character per row.

```
  <description>The number of characters in one row of this field.</description>
```

Is sdl sending a line more than 500 chars long in a single row?

madhuy, 2016-03-12:
Yes. mobile app sending more than 500 chars for each line. SDL core sending to HMI.

madhuy, 2016-03-14:
It is our Mobile app issue. Sorry for confusion.

closed 2016-03-14 by madhuy