smartdevicelink/sdl_core #444: Coverity for Code Coverage Reports and CI

justinjdickow, 2016-03-29:
We should consider integrating with Coverity, which scans open source projects for free and provides reports:  https://scan.coverity.com/

LuxoftAKutsan, 2016-06-14:
@justinjdickow  @JackLivio hello, what status of this issue?  I can see (coverity)[https://github.com/smartdevicelink/sdl_core/tree/coverity] branch with come yours commits.

LuxoftAKutsan, 2016-06-15:
@justinjdickow @JackLivio  is there are any updates? 
We wan't to use static code analyse on SDL. And I am searching for best solution, coverity looks rather good. Is there are any obstacles  to use coverity?

LuxoftAKutsan, 2016-06-16:
@justinjdickow @JackLivio is there are any updates?

LuxoftAKutsan, 2016-06-21:
@justinjdickow @JackLivio is there are any updates?  
I successfully used coverity for my fork from sdl https://scan.coverity.com/projects/luxoftakutsan-sdl_core 
I need admin rights for using coverify on sdl.

justinjdickow, 2016-06-21:
@LuxoftAKutsan we'll add you to the coverity project so you can review issues.

closed 2016-06-21 by justinjdickow

justinjdickow, 2016-06-21:
@LuxoftAKutsan Please let @JackLivio and I know if you intend on changing anything before you do.

LuxoftAKutsan, 2016-06-21:
@justinjdickow we would like to process static code analyse on develop branch weakly. 
For that purpose we will use specific target on build server and build server user https://github.com/JenkinsSDLOnCloud Will need access to submit static code analyse for project https://scan.coverity.com/projects/smartdevicelink-sdl_core

justinjdickow, 2016-06-21:
We are not okay with creating a dependency between our project and Luxoft owned CI servers, we'll have to discuss alternatives to this approach.

LuxoftAKutsan, 2016-06-21:
@justinjdickow Unfortunately we can't use free travis for it because collecting coverage is to long. 
I will investigate other possibilities for weakly coverage collecting. If you have any  ideas, please share.

LuxoftAKutsan, 2016-06-21:
@justinjdickow another way is to submit build manually once a weak, but I don't like this way.

LuxoftAKutsan, 2016-06-22:
@justinjdickow  we cant use free TravisCI for code coverage collecting because of time restrictions. 
We can use paid Travis CI. Also we can use Jenkins. Please note that analyse and reports will be available on scan.coverity.com Jenkins will perform collecting only collecting required artefacts for static code analyse and send to coverity. No  links will refer to Jenkins from open source repository.  
So can we use jenkins for triggering weakly static code analyse?