smartdevicelink/sdl_core #452: sometimes SDL core exits after "OnAppUnregistered" RPC

madhuy, 2016-03-31:
code base : master branch 
Commit number : e76f8e79294b55a58889ca9b6f90027918e7e25d

Logs:

INFO  [01 Jan 2001 02:44:22,526][ApplicationManager] HMICommandFactory::CreateCommand function_id: 21					
INFO  [01 Jan 2001 02:44:22,526][ApplicationManager] HMICommandFactory::CreateCommand request					
DEBUG [01 Jan 2001 02:44:22,526][ApplicationManager]  app_id << 1167829970Found = 0					
INFO  [01 Jan 2001 02:44:22,527][ApplicationManager] Attached schema to message, result if valid: 1					
INFO  [01 Jan 2001 02:44:22,527][ApplicationManager] Message to convert					
INFO  [01 Jan 2001 02:44:22,527][ApplicationManager] Message with protocol: 1					
INFO  [01 Jan 2001 02:44:22,528][ApplicationManager] Convertion result: {					
   "jsonrpc" : "2.0",					
   "method" : "BasicCommunication.OnAppUnregistered",					
   "params" : {					
      "appID" : 1167829970,					
      "unexpectedDisconnect" : true					
   }					
}					
					
INFO  [01 Jan 2001 02:44:22,528][ApplicationManager] Successfully parsed smart object into message					
INFO  [01 Jan 2001 02:44:22,528][ApplicationManager] Received message to hmi					
INFO  [01 Jan 2001 02:44:22,530][HMIMessageHandler] HMIMessageHandlerImpl::~sendMessageToHMI()					
INFO  [01 Jan 2001 02:44:22,531][ApplicationManager] Message to hmi given away.					
INFO  [01 Jan 2001 02:44:22,531][HMIMessageHandler] MessageBrokerAdapter::sendMessageToHMI					
DEBUG [01 Jan 2001 02:44:22,531][RequestController] app_id : 65537Requests waiting for execution count : 0Requests waiting for response count : 1					
DEBUG [01 Jan 2001 02:44:22,531][RequestController] app_id: 65537Waiting for execution0					
DEBUG [01 Jan 2001 02:44:22,531][RequestController] Waiting for execution 0					
DEBUG [01 Jan 2001 02:44:22,531][RequestController] Waiting for response count : 1					
DEBUG [01 Jan 2001 02:44:22,532][RequestController] Sleep for 11 secs					
DEBUG [01 Jan 2001 02:44:22,535][Utils] Timeout reset force: 11					
INFO  [01 Jan 2001 02:44:22,536][ApplicationManager] DeleteFile file /data/tsl/sdl/storage/SyncProxyTester584421907/icon.png					
INFO  [01 Jan 2001 02:44:22,537][ApplicationManager] DeleteFile file /data/tsl/sdl/storage/SyncProxyTester584421907/turn_forward.png					
INFO  [01 Jan 2001 02:44:22,537][ApplicationManager] DeleteFile file /data/tsl/sdl/storage/SyncProxyTester584421907/turn_right.png					
INFO  [01 Jan 2001 02:44:22,537][ApplicationManager] DeleteFile file /data/tsl/sdl/storage/SyncProxyTester584421907/action.png					
INFO  [01 Jan 2001 02:44:22,538][ApplicationManager] DeleteFile file /data/tsl/sdl/storage/SyncProxyTester584421907/turn_left.png					
DEBUG [01 Jan 2001 02:44:22,538][Utils] TimerThread is to be destroyed AudioStreamSuspend					
DEBUG [01 Jan 2001 02:44:22,540][Utils] Stopping thread #0 "AudioStreamSuspend "					
DEBUG [01 Jan 2001 02:44:22,540][Utils] Stopped thread #0 "AudioStreamSuspend "					
DEBUG [01 Jan 2001 02:44:22,540][Utils] Stopping thread #0 "AudioStreamSuspend "					
DEBUG [01 Jan 2001 02:44:22,540][Utils] Stopped thread #0 "AudioStreamSuspend "					
FATAL [01 Jan 2001 02:44:22,540][appMain] Stopping application due to segmentation fault

JackLivio, 2016-04-05:
@madhuy 

Are you seeing this over a single connection type of multiple?

Are you initiating the OnAppUnregistered?

madhuy, 2016-04-05:
@JackLivio 
#1 - Single connection
#2 - OnLanguageChange from Mobile or exit mobile app .

justinjdickow, 2016-04-11:
@madhuy Which connection type are you seeing this on? We're trying it on Android over WiFi and don't see a crash even though we can see an unexpected disconnect

madhuy, 2016-04-11:
we are using BT connection.

justinjdickow, 2016-04-11:
Is there a particular way you're forcing an unexpected disconnect? We see unexpected disconnect when turning off bluetooth as well as swiping the app away on the Android phone but have still not witnessed this crash.

madhuy, 2016-04-12:
OnLanguageChange RPC causes "OnAppUnregistered"

madhuy, 2016-04-12:
if it helps, below is the environment, compiler, tools chain been is used

    export EXTRA_CXX_FLAGS="-march=armv7-a -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a7 -pipe -feliminate-unused-debug-types";
    export TOOLCHAIN=/opt/poky/1.6.1/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-;
    export EXTERNALS=/opt/poky/1.6.1/sysroots/cortexa7hf-vfp-neon-poky-linux-gnueabi;

madhuy, 2016-04-14:
I see some fixes related timer in this https://github.com/smartdevicelink/sdl_core/pull/473/files. could you please check if this PR fixes it ?

justinjdickow, 2016-04-18:
@madhuy I can't check if that fixes the issue because I can't replicate it in the first place. Can you check if it fixes your issue?

madhuy, 2016-04-20:
It is crashing in pthread_join() method .

Program terminated with signal 11, Segmentation fault.
#0  0xb51164bc in pthread_join () from /home/sdlserver/Desktop/pio/binary/lib/libpthread.so.0
(gdb) bt
#0  0xb51164bc in pthread_join () from /home/sdlserver/Desktop/pio/binary/lib/

Could you please let us if you have any resolution for this.

madhuy, 2016-04-21:
I found the reason for crash . pthread_join() behavior is undefined when try to join uncreated thread.
in case of "AudioStreamSuspend" or "VideoStreamSuspend" timer threads, thread is not created (pthread_create() not been called) until WakeUpStreaming(https://github.com/smartdevicelink/sdl_core/blob/ea2eaa6a2ac6b4ce59aa3f8e89086aaff1db08b0/src/components/application_manager/src/application_impl.cc#L475). 

 ~ApplicationImpl() causes to  join(pthread_join) threads audio_stream_suspend_timer_ and video_stream_suspend_timer_ .  since no thread created (pthread_create()) it is crashing in arm-poky-linux-gnueabi platform.

to fix this i added check as below in https://github.com/smartdevicelink/sdl_core/blob/ea2eaa6a2ac6b4ce59aa3f8e89086aaff1db08b0/src/components/utils/src/threads/posix_thread.cc#L275

  if(handler_){
    pthread_join(handle_, NULL);
  }

justinjdickow, 2016-05-02:
@madhuy Can you submit a pull request for the fix into master called hotfix/452?

madhuy, 2016-05-18:
Created PULL request
https://github.com/smartdevicelink/sdl_core/pull/549

closed 2016-08-09 by madhuy