smartdevicelink/sdl_core #455: Alert timeout in case KEEP_CONTEXT button

madhuy, 2016-04-02:
SDL core resets 0 when alert has buttons.
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/alert_request.cc#L80

How do HMI should handle KEEP_CONTEXT type buttons?
HMI does call the OnResetTimeout but SDL core renew timeout with 0
https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/alert_request.cc#L131

Please clarify the behavior in case KEEP_CONTEXT

justinjdickow, 2016-04-11:
Right now the following use cases are possible

1. Alert has no soft buttons and times out -> Good
2. Alert has a KEEP_CONTEXT && (DEFAULT_ACTION || STEAL_FOCUS) -> Good, closes when one of the two are pressed, KEEP_CONTEXT button press sends a button event to mobile but alert stays open.
3. Alert has only DEFAULT_ACTION and STEAL_FOCUS soft buttons -> Good, closes when any buttons are pressed
4. Alert only has a KEEP_CONTEXT button -> **bad** -> alert never times out.

Seems like a bug, agree?

AnastasiyaBritanova, 2016-04-12:
Setting timeout to 0 - **is not a bug**.
- SDL "disables" internal timeout for **Alert with SoftButtons**. 
- Meaning - such Alert-w-SB display time should be dependent on "HMI-default-timeout-for-system-alerts", which might be rather different between HMIs and is definitely longer than SDL's default 10s.
- So. to "save resources" for HMI and to not make it send OnResetTimeout every 10s for such Alerts - SDL discards internal timeout in this particular case.

AnastasiyaBritanova, 2016-04-12:
about the case where KEEP_CONTEXT is the only button - it's not so bad in fact )
HMI can:
1) also provide own "Close" button -> and close with "ABORTED" result, for example 
2) close such dialog by HMI timeout with returning "TIMED_OUT" result.

closed 2016-04-12 by justinjdickow