smartdevicelink/sdl_core #460: Bug - SubscribeButton - CUSTOM_BUTTON

jgrix, 2016-04-06:
Cannot subscribe to CUSTOM_BUTTON after it has been unsubscribed to

-Reproduction Steps
1. Make sure CUSTOM_BUTTON is currently unsubscribed
2. Subscribe to CUSTOM_BUTTON 

-Expected Behavior
Successful response, button is subscribed

-Observed Behavior
The failure response is UNSUPPORTED_RESOURCE. Custom buttons were not subscribed to.

-OS & Version Information
Phone Type/Model and OS version: Samsung Note 5 with Android version 5.1.1
Mobile Application name and version: SyncProxyTester - 20160316-Nightly-Android
connection type: Wifi/Bluetooth/USB: USB
SDL_Core version: 4.0.9
HMI: Web based

AnastasiyaBritanova, 2016-04-13:
please make sure CUSTOM_BUTTON is returned as a supported buttonName in Buttons.GetCapabilities_response from HMI 
(alternatively - check RegisterAppInterface_response ("ButtonCapabilities") - it must list CUSTOM_BUTTON for expected successful scenario).

yang1070, 2016-04-13:
yes, we have CUSTOM_BUTTON
with shortPressAvailable/longPressAvailable/upDownAvailable all true in the
Buttons.GetCapabilities response, but sdl core still do the same.

Thanks,
Zhimin

jgrix, 2016-04-13:
Please note this behavior is happening with the web based HMI from  https://github.com/smartdevicelink/sdl_hmi

jacobkeeler, 2016-09-15:
@jgrix @yang1070 This appears to be an issue with the HMI having implemented something that isn't yet in Core.

To start, it looks like this is happening because the ButtonCapabilities are not actually being updated with the Button.GetCapabilities response from the HMI. This is because Core thinks that the response that it received is incorrectly formed.

The reason for this is that the HMI includes the PLAY_PAUSE button in its Button.GetCapabilities response, which is not yet in the HMI API used by Core (there is currently a pull request for this here: #766). Since it is not included in the API, the message parser marks it with an INVALID_DATA response.

The easiest way to fix this issue as it stands is to remove the PLAY_PAUSE button from the HMI capabilities [here](https://github.com/smartdevicelink/sdl_hmi/blob/master/ffw/ButtonsRPC.js#L243-L246) and [here](https://github.com/smartdevicelink/sdl_hmi/blob/master/ffw/UIRPC.js#L799-L802).

closed 2017-04-18 by jacobkeeler