smartdevicelink/sdl_core #462: How to solve this error:cc1plus: all warnings being treated as errors

TaoheGit, 2016-04-07:
[ 26%] Linking CXX static library libProtocolHandler.a
[ 26%] Built target ProtocolHandler
Scanning dependencies of target connectionHandler
[ 26%] Building CXX object src/components/connection_handler/CMakeFiles/connectionHandler.dir/src/connection_handler_impl.cc.o
/home/taohe/SmartDeviceLink/sdl_core/src/components/connection_handler/src/connection_handler_impl.cc: In member function ‘void connection_handler::ConnectionHandlerImpl::OnConnectionEnded(const ConnectionUID&)’:
/home/taohe/SmartDeviceLink/sdl_core/src/components/connection_handler/src/connection_handler_impl.cc:923:8: error: ‘template<class> class std::auto_ptr’ is deprecated [-Werror=deprecated-declarations]
   std::auto_ptr<Connection> connection(itr->second);
        ^
In file included from /usr/include/c++/5.3.0/memory:81:0,
                 from /home/taohe/SmartDeviceLink/sdl_core/src/components/connection_handler/src/connection_handler_impl.cc:37:
/usr/include/c++/5.3.0/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^
cc1plus: all warnings being treated as errors
src/components/connection_handler/CMakeFiles/connectionHandler.dir/build.make:62: recipe for target 'src/components/connection_handler/CMakeFiles/connectionHandler.dir/src/connection_handler_impl.cc.o' failed
make[2]: *** [src/components/connection_handler/CMakeFiles/connectionHandler.dir/src/connection_handler_impl.cc.o] Error 1
CMakeFiles/Makefile2:1499: recipe for target 'src/components/connection_handler/CMakeFiles/connectionHandler.dir/all' failed
make[1]: *** [src/components/connection_handler/CMakeFiles/connectionHandler.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

BrandonHe, 2016-04-08:
The warning-suppression does not work with gcc-5.3.0. Suggest you use 4.8.4.
**Or** you can change your local sdl_core code in [connection_handler_impl.cc#L923] (https://github.com/smartdevicelink/sdl_core/blob/master/src/components/connection_handler/src/connection_handler_impl.cc#L923), as to below:
``` 
std::unique_ptr<Connection> connection(itr->second);
```
because auto_ptr is deprecated in C++11

malirod, 2016-04-08:
@TaoheGit, the problem occurs because gcc5 uses c++11 by default and auto_ptr, indeed, deprecated there. Unfortunately we cannot use c++11 features(unique_ptr is one of them), because we have to support older compilers.
Propose two options:
- use pre gcc5 compiler. e.g. gcc 4.9
- set c++98 explicitly for the gcc5. This should be possible with something like -std=c++98 (or -std=c++03, not sure about flags correctness)

TaoheGit, 2016-05-05:
@malirod @BrandonHe 
Thanks

closed 2016-05-05 by TaoheGit