smartdevicelink/sdl_core #466: onFindApplication Notification Not implemented in SDL core

madhuy, 2016-04-07:
https://github.com/smartdevicelink/sdl_core/blob/ea2eaa6a2ac6b4ce59aa3f8e89086aaff1db08b0/src/components/application_manager/src/commands/hmi/on_find_applications.cc#L48

Please clarify if we need this API or not .

and also, what does "Find applications" in ford TDK does?

justinjdickow, 2016-04-12:
Historically, the RPC was going to be used to search for SDL Enabled applications on a connected bluetooth device. Currently, SDL searches for bluetooth devices and applications [periodically](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/transport_manager/src/bluetooth/bluetooth_device_scanner.cc#L379-L383) without interaction from the user. "Find Applications" on Sync doesn't appear to interact with SDL.

I will create another ticket to deprecate and remove this API

closed 2016-04-12 by justinjdickow

madhuy, 2016-04-12:
AS I mentioned in #469 "Find Applications" in TDK does some thing to show applications.