smartdevicelink/sdl_core #469: After BT disconnecting and connecting SPP is not established with Android Multiplexer based apps

madhuy, 2016-04-08:
Here is test environment
1. Have 2 simple multiplexer based apps.
2.using SDL core version 4.0.10
3. Android phone and model : nexus 6 and version 5.1.1

Repro steps
1. start sdl core
2. all application shows in HMI and function properly
3. Disconnect BT in Mobile(settings->Bluetooth)
4. Connect BT
5. Apps are not shown on HMI. Connection establishment failed.

Notes: I tried with SYNC TDK and the behavior is different.
 in TDK, some times apps shows and communicates properly. Some time I have to select "Find Applications" Button to establish connection.

I will attach logs.

madhuy, 2016-04-08:
[log_android_multiplexer.txt](https://github.com/smartdevicelink/sdl_core/files/210912/log_android_multiplexer.txt)
[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/210914/SmartDeviceLinkCore.txt)

madhuy, 2016-04-11:
Attaching logs from HelloSDL app . also attaching Mobile apps to repro issue. 
[log2.txt](https://github.com/smartdevicelink/sdl_core/files/213692/log2.txt)

[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/213694/SmartDeviceLinkCore.txt)
[apps.zip](https://github.com/smartdevicelink/sdl_core/files/213698/apps.zip)

madhuy, 2016-04-11:
And also, I verified with TDK with latest build. 1 out of 4 times, user have to select "find applications" button to show apps after disconnecting and connecting BT.

MprenticeUIE, 2016-04-25:
Repros on Nexus 5

justinjdickow, 2016-07-12:
@joeygrover Is this something we should move to the sdl_android repo?

joeygrover, 2016-07-12:
Is this issue still valid since multiplexing was merged into dev in the sdl_android repo?

jthrun, 2016-08-05:
I still see this happening on both the TDK and our dev system using the generic UUID build of SDLCore.  It seems like older SDL apps (like Pandora installed from Google Play store) don't have the problem.  That is, switching BT off then back on, Pandora icon is still shown in launcher and can be run but multiplexer apps aren't available.  Happens even if there isn't an older SDL app on the handset.

joeygrover, 2016-08-18:
Can you try this on a device running Android 6.x+? I will look into getting a device with an older Android OS either way. 

Also, what version of software is running on the TDK?

jthrun, 2016-08-18:
TDK says: SYNC 3.0, Software Version 1.0, Build 15139

looking for a 6.x device...

joeygrover, 2016-08-18:
Thank you. A final note, you can still get apps to connect if you press "find new apps" right? It just doesn't happen automatically sometimes on the TDK?

madhuy, 2016-08-24:
But we are not sure what exactly, "Find New applications" does ? I was assuming this calls OnFindApplications but OnFindApplications is [deprecated](https://github.com/smartdevicelink/sdl_core/issues/480) .  Could you please provide me what exactly "Find New app" does in HMI level or SDL core level.

joeygrover, 2016-08-24:
@stefanek22 @JackLivio can you advise what happens between HMI and Core when this happens? Or even on the TDK?

jthrun, 2016-08-26:
@joeygrover I finally found a Nexus 6 with v6.0.1.  Tested with the TDK and after restarting BT on the device, the apps showed up, then went away and came back a couple times, then things stabilized and both apps were shown.  I was using HelloSdl as one of the apps and, after BT restart I had trouble launching it, got an error about "can't start the app" until I launched/foregrounded the app on the handset, then no problems.

jthrun, 2016-09-30:
I can not reproduce this using 4.1rc1 of SDL Android and the 4.1 of SDL Core.  Tried both in dev environment and on TDK (older SDL Core, but using 4.1rc1 of SDL Android seems to resolve the issue).  However, on dev environment, if a non-multiplexing SDL app makes connection first then no multiplexing apps show up.  I think this is not an issue for us since our build of SDL Core will only connect to the companion app.  @joeygrover I think this issue can be closed.

closed 2016-10-03 by madhuy