smartdevicelink/sdl_core #479: AddCommand API responses GENERIC_ERROR response after Sending  200 addCommand

madhuy, 2016-04-11:
We are trying to test with 200 AddCommand.

We noticed,
1. for first140 requests SDL core responses SUCCESS
2. remaining 60 Requests it reponse GENERIC_ERROR. but it added to command list and shows in HMI.
Note: We are using default time out 20 sec.

Please let us know if we are doing anything wrong.

Attaching logs.
[SmartDeviceLinkCore-2.txt](https://github.com/smartdevicelink/sdl_core/files/213913/SmartDeviceLinkCore-2.txt)

madhuy, 2016-04-12:
in iOS, SDL core responds multiple SUCCESS responses for same AddCommand Request. it is reproducible with Wifi Transport also.

MprenticeUIE, 2016-04-18:
The issue appears to be tied to timing (SDL core takes to process and respond) not a specific request limitation.

JackLivio, 2016-04-20:
@madhuy @MprenticeUIE 

So the issue here its that the AddCommands are timing out. Since there are so many add commands being sent by mobile at once, it doesn't appear that core is able to process all 200 commands before hitting the default timeout. 

By default this value is set 10 seconds in the smartDeviceLink.ini file. Once possible fix is to increase this timeout value. 

Another suggestion would be to try and throttle how fast the mobile device is sending the add commands.

JackLivio, 2016-04-20:
Additionally I am looking into if the request controller should be updating its timers under certain circumstances. This might be a case.

madhuy, 2016-04-20:
@JackLivio  the default time out is 20 sec. 
"Note: We are using default time out 20 sec"

MprenticeUIE, 2016-05-20:
Is there any evidence that an app would add more then 140 commands?

madhuy, 2016-05-24:
We experimented below as per Ford Recommendation in their Developer portal, It is working. 

_Is there a limit to how fast I can send AddCommand requests?
If you need to send a large number of requests to set up your application up when leaving HMI Level of NONE, the following is recommended:
Send a small number of the most important requests immediately. These usually include button subscriptions, custom prompts (SetGlobalProperties), and four or five of the most important commands.
Using a timer or background thread, send the remaining AddCommand requests in batches of 5 or 10 every second or so.
The most AddCommands any application has sent at startup is around 200, which takes about 10-15 seconds. Commands sent first will be available as voice commands first. So, it’s recommended to send more important commands first. It is technically possible to send more than 200 commands, but this will increase loading time for the application and decrease SYNC®’s response time to user input._

madhuy, 2016-08-04:
Some more points
1. we tried in TDK with Android device , TDK also returns GENERIC_ERROR for some requests
2. if AddCommand has VR Coomands , Mobile app gets SUCCESS response though it take some time
3. if AddCommand dont have any VR commands , Mobile app gets GENERIC_ERROR for few requests.  if we increase Timeout Value to 75 sec all of AddComand request gets Success.

Attaching Logs for same
AddCommand With out VR commands :

[SDLCore_addcmd200_without_vr_defaulttimeout20.log.zip](https://github.com/smartdevicelink/sdl_core/files/402733/SDLCore_addcmd200_without_vr_defaulttimeout20.log.zip)

AddCommands with  VR command : 
[SDLCore_addcmd200_with_vr_defaulttimeout20.log.zip](https://github.com/smartdevicelink/sdl_core/files/402741/SDLCore_addcmd200_with_vr_defaulttimeout20.log.zip)

BrandonHe, 2016-08-05:
@madhuy @JackLivio We also find this GENERIC_ERROR issue caused by time_out, optional solution is to change the configuration in smartdevicelink.ini, but that's a workaround way. I sent a PR #660, and try to fix this issue by unblocking the waiting threads in RequestController. As @LuxoftAKutsan required, I will create Unit Test for this PR.

madhuy, 2016-09-23:
@BrandonHe is your PR #660 merged to 4.1?

madhuy, 2016-10-03:
it looks like #660 is closed with out merging. We changed to 60 sec time out as workaround it works . it is work around. what is the ETA on this ?

BrandonHe, 2016-10-04:
@madhuy #660 is not merged to 4.1, because @LuxoftAKutsan said it has same logic with previous code in ```RequestController```,but I do add thread wait in ```RequestController::onTimer()``` to make request process not time out and passed ATF test [ATF_AddSubMenu_fix_test_result.txt](https://github.com/smartdevicelink/sdl_core/files/342777/ATF_AddSubMenu_fix_test_result.txt), maybe you can take some time to change the code same with [modification](https://github.com/smartdevicelink/sdl_core/pull/660/files) in this #660 to have a test, to take a look at if it's also useful in your code.

madhuy, 2016-10-04:
@BrandonHe I can test. but is this changes going to merge in future with separate PR?

BrandonHe, 2016-10-04:
@madhuy Nope, if this change also effective for your code, I can reopen to ask for merging.

JackLivio, 2016-10-10:
@BrandonHe Wanted to bump this issue. Were you going to revisit your PR?

madhuy, 2016-10-12:
@BrandonHe after integrating  PR #660 . AddCommand responses properly. 
@JackLivio Please prioritize #660

madhuy, 2016-10-18:
@BrandonHe Could you please prioritize the #660 PR?

closed 2016-10-24 by jacobkeeler

joeygrover, 2016-10-24:
Fixed with #909

PR #909 (2016-10-24): Hotfix/resolve atf add sub menu known issues
Replace PR #660, to resolve #479 and #621.
Please help to review @jacobkeeler @LuxoftAKutsan @AGaliuzov @madhuy 

According to requirements, change PR name to  **hotfix** and branched off **master**, have rebased and squashed the original commits. 
For detail information, please reference PR #660.