smartdevicelink/sdl_core #491: Microphone format not set, settings from PerformAudioPassThru request ignored by OnAudioPassThru notification

chrisdrehernuance, 2016-04-14:
The OnAudioPassThru notification sends microphone audio data from the HU to the mobile device as binary PCM data (preceded by a wav file header).  Unfortunately, this data does not match the PCM format requested in the preceding PerformAudioPassThru request from the phone.

For example, PerformAudioPassThru can request microphone audio data in a format of 8-bit PCM at 22KHz but OnAudioPassThru will return data in the OS's default format (ex: 16-bit 44KHz).

Recommend updating the file from_mic_to_file_recorder_thread.cc to set the sampling Hz, the bits-per-sample, and number of channels.

LitvinenkoIra, 2017-12-01:
Priority Low.
Offer for enhancement, not a defect.

JackLivio, 2018-03-01:
@chrisdrehernuance That audio format might not be supported by the head unit. Please check the audioPassThruCapabilities struct in the Register App Interface response. Also having multiple supported audioPassThruCapabilities is an accepted proposal. Please refer to this proposal which will be implemented into core in the future. https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0138-hmi-audiopassthru-capability.md

ShobhitAd, 2020-06-30:
Closed via #2184

closed 2020-06-30 by ShobhitAd