smartdevicelink/sdl_core #492: Heartbeat starvation during AudioPassThru over Bluetooth, resulting in SDL apps dropped by SDL Core

chrisdrehernuance, 2016-04-15:
While using OnAudioPassThru over Bluetooth, heartbeat starvation has intermittently occurred resulting in the SDL Core dropping SDL apps from the mobile phone.

The current working theory is that:

1. Bluetooth is bandwidth constrained.  At 3Mbps, Bluetooth offers a max of 2.1Mbps to applications (after subtracting overhead) but only under ideal conditions.
2. Due to issue #491, PCM audio data can consume up to 0.7Mbps (16-bits at 44.1KHz).
3. Theoretically, a heartbeat request could be treated as high priority packet that jumps to the head of a packet queue, skipping ahead of any temporarily delayed OnAudioPassThru packets.  However, the code in `bool ThreadedSocketConnection::Send()` in file `threaded_socket_connection.cc` only uses a normal queue instead of a PrioritizedQueue.  If it receives a burst of OnAudioPassThru packets, then a heartbeat packet that arrives a moment later will not be moved to the head of the queue.  Note: we do see that `ProtocolHandlerImpl` uses a `PrioritizedQueue` but suspect that the packets are moving in and out of the queue quickly before they eventually end up in a normal queue where most of the backlogged packets reside.

AnastasiyaBritanova, 2016-04-25:
HeartBeat (HB) request is not treated as a high priority request by SDL -> because mobile app should reset its "heartbeat expectation" timer by *any* data received from SDL:

app -> get HB_request *or* RPC *or* frame(s) ==> reset "HB expectation" timer.

chrisdrehernuance, 2016-04-25:
I wasn't very clear above.  The SDL Core on the head unit is the side whose heartbeat expectation timer is timing out, not the the mobile app side.

When the head unit is sending microphone audio to the mobile app, it is sent as `OnAudioPassThru` notifications from the head unit to the mobile app.  These notifications are consumed by the mobile app and do not require the mobile app to response back to the head unit.  From the head unit's perspective, data is flowing out but no data is received in response.  This is normal in the current design of AudioPassThru.  Thus, the head unit will periodically generate a heartbeat packet to confirm the mobile app is still responsive.  However, because the output queue gets backed up and the heartbeat is not able to jump the queue in `ThreadedSocketConnection `(though it can in `ProtocolHandlerImpl`'s `PrioritizedQueue`), the delay causes SDL Core on the head unit to timeout the SDL app.

AnastasiyaBritanova, 2016-04-26:
understood. thank you for explaining.

From the requirements, 
> •	**Both the mobile proxy** and core need to have the capability to send a heartbeat and respond to a heartbeat.
> •	**Each side is responsible** for letting the other side know it’s there.

so, if SDL queued the HB (per scenario above) - mobile app has own HB_request to let SDL know it's there.

chrisdrehernuance, 2016-04-26:
Correct.

The last post does bring up an alternative solution.  The mobile side's HB code could be modified to send out a HB_request if the mobile side has not _transmitted_ anything in a while to let the SDL Core know that the mobile app is still there.  This logic would be separate from the mobile app's own timeout logic if it has not _received_ a HB request/response, RPC, or frame from the SDL Core.  The downside to this solution is that the mobile app does not know what the duration the SDL Core has for its timeout (i.e. does the mobile app generate a HB request after 1 second, 5 seconds, or 30 seconds of no transmissions).

LitvinenkoIra, 2017-12-01:
Priority Low.
Logic of HB was changed according to last comment in issue, so there is big probability that defect is not actual.

jacobkeeler, 2018-03-02:
@chrisdrehernuance Are you still experiencing this issue on the latest release? Based on the comment from @LitvinenkoIra, this may already be fixed.

jacobkeeler, 2018-03-19:
@chrisdrehernuance Closing, reopen if you are still experiencing this issue.

closed 2018-03-19 by jacobkeeler