smartdevicelink/sdl_core #495: Register App Interface Response should give information relating to drawable area.

asm09fsu, 2016-04-16:
For video streaming applications, there is no way for developers to know what the drawable area is. 

We receive the screen dimensions for the connected head unit, however we cannot assume that all head units in the future will have a drawable area equivalent to the size of the screens dimensions.

We should ideally return a `drawable_rect` property or something with an equivalent name. This should return an inset rect for head units. For example: screens dimensions may be 600x400, but a drawable rect of {0, 60, 600, 300}.

LitvinenkoIra, 2017-12-01:
Priority Low.
Enhancement, not a defect, not affect SDL functionality.

JackLivio, 2018-03-01:
I believe the ScreenParams struct supplied by the display capabilities struct in the register app interface response is supposed to be the size of the screen dedicated to sdl for video streaming within the head unit ui based off of the parameter description. 

```
        <param name="screenParams" type="ScreenParams" mandatory="false">
            <description>A set of all parameters related to a prescribed screen area (e.g. for video / touch input).</description>
        </param>
```

Also in the case that you mentioned where the video only plays on a  smaller part of the screen, this case is covered by the preferredResolution parameter in the VideoStreamingCapability struct. 

```
    <struct name="VideoStreamingCapability">
        <description>Contains information about this system's video streaming capabilities.</description>
        <param name="preferredResolution" type="ImageResolution" mandatory="false">
            <description>The preferred resolution of a video stream for decoding and rendering on HMI.</description>
        </param>
        <param name="maxBitrate" type="Integer" minvalue="0" maxvalue="2147483647" mandatory="false">
            <description>The maximum bitrate of video stream that is supported, in kbps.</description>
        </param>
        <param name="supportedFormats" type="VideoStreamingFormat" array="true" mandatory="false">
            <description>Detailed information on each format supported by this system, in its preferred order (i.e. the first element in the array is most preferable to the system). Each object will contain a VideoStreamingFormat that describes what can be expected.</description>
        </param>
        <param name="hapticSpatialDataSupported" type="Boolean" mandatory="false">
            <description>True if the system can utilize the haptic spatial data from the source being streamed. If not included, it can be assumed the module doesn't support haptic spatial data'. </description>
        </param>
    </struct>
```

If you feel that this solution still does not address your issue please submit an evolution proposal to the sdl_evolution repository.

closed 2018-03-01 by JackLivio