smartdevicelink/sdl_core #518: Compile failed with -DHMI=qt -DHMI2=ON -DEXTENDED_MEDIA_MODE=ON on new DialNumber RPC

chinlin0924, 2016-04-28:
/autooem/projects/SDL/sdl_core/src/components/application_manager/src/commands/hmi/dial_number_response.cc: In member function ‘virtual void application_manager::commands::hmi::DialNumberResponse::Run()’:
/autooem/projects/SDL/sdl_core/src/components/application_manager/src/commands/hmi/dial_number_response.cc:48:29: error: ‘BasicCommunication_DialNumber’ is not a member of ‘hmi_apis::FunctionID’
   event_engine::Event event(hmi_apis::FunctionID::BasicCommunication_DialNumber);
                             ^
make[2]: *** [src/components/application_manager/CMakeFiles/AMHMICommandsLibrary.dir/src/commands/hmi/dial_number_response.cc.o] Error 1
make[1]: *** [src/components/application_manager/CMakeFiles/AMHMICommandsLibrary.dir/all] Error 2
make: *** [all] Error 2

jacobkeeler, 2017-10-19:
See https://github.com/smartdevicelink/sdl_core/issues/1824 for more details about the QT HMI. The bottom line is that it hasn't been supported in a long time, and it is not yet clear whether we will in the future. I hope to have more details sometime in November.

LitvinenkoIra, 2017-12-01:
Priority Low.
Related to Qt HMI

JackLivio, 2018-03-01:
@chinlin0924 There have been updates to how the QT HMI and dbus are built in the project in the new release candidate. Please give release/4.5.0 a try. You will need to install libdbus as a dependency. Closing issue to being outdated.

closed 2018-03-01 by JackLivio

chinlin0924, 2018-03-02:
ok! looks great thanks