smartdevicelink/sdl_core #528: How to run sdl_core in an android board

TaoheGit, 2016-05-05:
I want to know how transplant sdl_core on an android board,I have no idear

BrandonHe, 2016-05-05:
Hi Tao,
We have transplanted sdl_core on android port, but now it runs on sdl_core 3.6 version. We will have a latest version release combined with sdl_core 4.1.0 version(it's also not released yet) in the future. You can directly download an available apk file from [here](https://github.com/APCVSRepo/hmi_sdl_android). I will add instruction wiki files for how to build on andriod board in these days.

Thanks!
Ford AP team

TaoheGit, 2016-05-05:
@BrandonHe 
Thank you very mach
I am looking forward to your good news.

TaoheGit, 2016-05-05:
@BrandonHe 
非常感谢能得到你的帮助，看到你们在git的项目我觉得很兴奋
希望能和你们多交流

closed 2016-05-09 by TaoheGit