smartdevicelink/sdl_core #572: ARM cross compilation issue

lbtianya, 2016-06-06:
1.x86 path  /usr/local/include and /usr/local/lib always appear in  flag.make and link.txt , arm-gcc 4.4. will thow a error on them,
...
"include location "/usr/local/include" is unsafe for cross-compilation"
"library search path "/usr/local/lib" is unsafe for cross-compilation"

God, I need to comment such lines every time.  how to remove it in embedded env, but use my RPATH setting or LDFLAG/CFLAG  env.

2.Some modes like , Apr/log4cxx , always find link path '/usr/lib" instead of my ARM RPATH.
...
/bin/grep: /usr/lib/libuuid.la: No such file or directory
/bin/sed: can't read /usr/lib/libuuid.la: No such file or directory
...
/bin/sed: can't read /usr/lib/libexpat.la: No such file or directory
libtool: link: `/usr/lib/libexpat.la' is not a valid libtool archive

so strange modules, if have any method to modify it from cmake script.

chenbd, 2016-08-14:
sdl core needs a little more work when cross compile, and current cmake build system is not well supported when cross compile, specially for modules like apr apr-util apache-log4cxx dbus libusb.

i guess sdl core should do some work for better support cross compile

jacobkeeler, 2018-03-01:
@lbtianya Are you still encountering this on the newest release?

jacobkeeler, 2018-03-08:
Closing for now, reopen if you are still experiencing this issue.

closed 2018-03-08 by jacobkeeler