smartdevicelink/sdl_core #620: CreateChoiceintercationSet with MAX param values - Performance improvement

madhuy, 2016-06-17:
We have test case CreateChoiceintercationSet with all params MAX values( as per spec) .  It is taking 50 sec to 60 sec to process and respond back to Mobile app.

since default time out is 20 sec. this always returns TIMEOUT error

Below is the JSON data to SDL core. it might be useful to profile 
[interactionChoiceSet_boundary_MAX.txt](https://github.com/smartdevicelink/sdl_core/files/320796/interactionChoiceSet_boundary_MAX.txt)

justinjdickow, 2016-07-07:
Is this happening on SDL Core or on a Ford TDK?

madhuy, 2016-07-13:
Yes, We are able to repro in FORD TDK as well.

jacobkeeler, 2016-09-07:
I was able to reproduce the issue on a TDK as well (responds with GENERIC_ERROR), but not when testing the same command using the recently merged 4.1 release. In Core 4.1, it still takes about 60 seconds for Core to fully receive the message from an Android bluetooth application, but it does not seem to time out (Core responds with SUCCESS).

I also tested the same command using the ATF, and the command only took around 2.5 seconds for SDL Core to process the message and respond. This implies to me that the transport itself may be the bottleneck, although Bluetooth should be able to send the message significantly faster in theory (the command is around 1.3MB).

@madhuy does your test case use Bluetooth as the transport?

madhuy, 2016-09-07:
@jacobkeeler Yes, our testcase uses Bluetooth transport

jacobkeeler, 2016-09-13:
@madhuy Are you still having the timeout issue with the new release? 

I have also now tested the same command with iOS/TCP, and that takes around 16 seconds for that to complete, and around 10 seconds accounted for by communication with the web HMI. Based on the logs, Core only took around 1/10th of a second to actually process the message. 

It seems to me like the Android Library might be the cause of the bottlenecking the Android/Bluetooth case.

In any case, even with the 60 second timing for Android, nothing seems to be sending a timeout anymore. If the timeout is no longer occurring on your side, then I would imagine that this would no longer be an issue.

madhuy, 2016-10-03:
with 60 seconds timeout it is not happening.

closed 2016-10-03 by madhuy