smartdevicelink/sdl_core #621: Scrollablemessage with DEFAULT_ACTION/STEAL_FOCUS buttons does not send onBUttonPress notification if mobile app send scrollbale message twice.

madhuy, 2016-06-17:
1. Send Scrollmessage with DEFAULT_ACTION/STEAL_FOCUS buttons
2. Click Button  DEFAULT_ACTION/STEAL_FOCUS  
3. send onBUttonPress notification to mobile app- works as expected.
4. send another scrollable message with DEFAULT_ACTION
5. Click Button  DEFAULT_ACTION/STEAL_FOCUS  
6. Does NOT send onBUttonPress notification to mobile app- NOT working.

Please let me know if you need any info.

AnastasiyaBritanova, 2016-07-08:
SDL logs would be helpful in this case - to identify who's responsible: HMI for not providing or SDL for not transferring these notifications.

justinjdickow, 2016-07-12:
@madhuy Please see Ana's comments and provide logs to this issue.

madhuy, 2016-07-13:
attaching SDL core logs
[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/360681/SmartDeviceLinkCore.txt)

jacobkeeler, 2016-07-19:
I've not been able to reproduce the problem on my end using the steps you gave, the OnButtonPress still went through to the mobile side on my end.

Based on the logs you uploaded, it appears that the HMI is sending the OnButtonPress to core correctly, as I see the log with the message from the HMI.

```
INFO  [05 Feb 2016 20:46:10,924][ApplicationManager] 			Message to convert: protocol 0; json {"jsonrpc":"2.0","method":"Buttons.OnButtonPress","params":{"appID":1845945318,"customButtonID":5508,"mode":1,"name":15}}
```

The issue may be that your app is not being subscribed to the buttons in the ScrollableMessage (or Core is not correctly keeping track of which buttons are subscribed).

```
ERROR [05 Feb 2016 20:46:10,926][Commands] virtual void application_manager::commands::mobile::OnButtonPressNotification::Run(): Application doesn't subscribed to this custom_button_id.
```

I'll keep looking through the logs and code to see if I can figure out why it isn't being subscribed to the buttons.

AnastasiyaBritanova, 2016-07-20:
@madhuy @jacobkeeler - please pay attention to the **order of messages**:
line 16471: SDL -> HMI : UI.ScrollableMessage with SBs id=5506 and id=5507 
line 17089: HMI -> SDL : UI.ScrollableMessage (SUCCESS) 
line 17156: SDL -> app : ScrollableMessage (SUCCESS, success:true)
line 17201: HMI -> SDL : Buttons.OnButtonPress (**id=5506**)
==>
-> it's not about wrong subscription status
-> it's about the associated RPC is already responded and removed from SDL's processing (SDL ignores OnButtonPress notification as referenced buttonID "does not exist").

jacobkeeler, 2016-07-20:
@madhuy Are you are using a separate HMI from the example HTML5 HMI on GitHub?

If that is the case, then this issue can be closed since it is an HMI problem. I am unable to replicate the same problem using the example HMI, the messages are sent in the correct order, so it doesn't appear to be an issue on this side.

madhuy, 2016-07-20:
@jacobkeeler  Thanks, for the analysis
Yes, We are using different HMI.

HMI sends message order correctly. but SDL core processing in different order. Even we tried sending after some delay from HMI . but still behavior is same.

AnastasiyaBritanova, 2016-07-21:
@madhuy 
from the logs it's seen that ScrollableMessage is immediately responded (like in 1 sec). 
Please note, it's expected HMI to respond only **after** the ScrollableMessage dialog is closed (either by timeout or by button press).

@jacobkeeler 
from the logs, SDL does not always send `"timeout"`which is mandatory for ScrollableMessage per HMI_API. That could be an issue (immediate response) if HMI does not apply a default timeout for popups.

jacobkeeler, 2016-07-21:
@AnastasiyaBritanova Where in the log are you seeing this? I don't see any messages from SDL which are missing the timeout field

In the log, I see four UI.ScrollableMessage messages sent to the HMI, and each of them have a timeout value of 30000

```
line 13051:      "timeout" : 30000
line 15186:      "timeout" : 30000
line 16497:      "timeout" : 30000
line 17777:      "timeout" : 30000
```

AnastasiyaBritanova, 2016-07-22:
@jacobkeeler the timeouts you're referencing - are SDL's internal timeouts set for ScrollableMessage. But in addition, SDL must send these timeouts **within json** message to HMI as well.
See:
Line 12942: json -> no "timeout" in the json message
Line 16397: json -> no "timeout" in the json  message
Line 17678: json -> no "timeout" in the json  message

The only exception (how the ScrMes json message should normally look like):
`Line 15077: INFO  [05 Feb 2016 20:45:17,801][ApplicationManager] 			Message to convert: protocol 2; json {"timeout":30000,"softButtons":[{....`

jacobkeeler, 2016-07-22:
@AnastasiyaBritanova First off, thank you for all of your help 😀 

Second, after looking through the code, it looks like the messages that you are referring to are the JSON messages which are **received** by SDL. The log you pointed out is in the ConvertMessageToSO function, which is only used for incoming messages from mobile or HMI. 

This means that the messages that you pointed are the requests from the mobile side, where the timeout field is not required, thus why some of them are missing this field.

The messages that I pointed out in my last comment show the exact JSON that is sent out by SDL (from what I could see), converted from the internal representation that you mentioned. The "convertion result: " message is used to show the result of the ConvertSOtoMessage function, which is only used for outgoing messages, and this result appears to be what is directly sent to the destination. For this reason, it does not appear to me that SDL is failing to send a timeout to the HMI.

However, what you pointed out made me realize something. I now see that the only ScrollableMessage in the log file in which the response messages were in the correct order was the one in which a timeout was sent from mobile. In other words, it doesn't appear that the issue is as simple as "SDL is not sending a timeout to the HMI", but the timeout is likely related to the issue.

@madhuy Could you try including a timeout field with all of the ScrollableMessage mobile requests from your mobile app, and see if that makes the responses come in the correct order, just to make sure this isn't a coincidence? It is definitely an issue if SDL has problems when the timeout field is omitted from the mobile request. 

In the meantime, I will try and look into whether the omission of the timeout field on the mobile side could cause such an issue on the SDL side.

jacobkeeler, 2016-07-27:
@madhuy Have you been able to determine whether including a timeout in the ScrollableMessage requests fixes the issue? 

I've been unable to find anything obvious in the SDL code that would affect the outcome if a timeout is omitted (vs. included with the default value of 30000). If including the timeout works, I will know the this is where the issue lies on the SDL side, because on the HMI side there shouldn't be a difference between the two cases. 

I am unable to reproduce the issue on this end, so I cannot test this myself.

joeygrover, 2016-08-08:
@madhuy have you been able to take a look at this?

madhuy, 2016-08-17:
@joeygrover we were able to reproduce this issue with 60 sec timeout and latest SDL core.

Attaching logs. 
[core93_sdl176.txt](https://github.com/smartdevicelink/sdl_core/files/423152/core93_sdl176.txt)

jacobkeeler, 2016-08-23:
@madhuy Alright, so based on the logs you sent, it doesn't appear that the manual inclusion of a timeout by the mobile app makes a difference, meaning that the successful cases in the previous log seem to be just a coincidence.

This implies to me that the issue is still with the HMI you are using, as using the sample HMI results in SDL Core consistently receiving the messages in the correct order. There doesn't seem to be any delay introduced by the sample HMI to force the messages to arrive in the correct order, either. 

Have you been able to reproduce this issue on a TDK? If you are able to do so, that would indicate that there is an issue with Core, and that there is just a difference in our setups.

madhuy, 2016-08-29:
@JackLivio in TDK, Scrollable messages shows only one button though mobile app send multiple buttons.

madhuy, 2016-08-29:
Here is HMI flow.
1) send onButtonPressNotifcation after user selecting Button on scrollable message 
2) Send UI.Scrollblemessage with ABORTED.

But, SDL core processing  
1) UI.Scrollblemessage, which causing [unsubscribe](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/scrollable_message_response.cc#L56) the buttons
2) onButtonPressNotifcation - since buttons are unsubscribed step 1. SDL core is not sending button notification to mobile app.

I am not sure why SDL core processing UI.scrollblemessage first and then onButtonPress notification

And also, why is buttons subscription is been handled differently in case of scrollable message compare to alert.

Please let me know if HMI is doing dome thing wrong in case of [scrollablemessage](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/application_impl.cc#L866) buttons.

madhuy, 2016-08-29:
I did experiment by  commenting below 
1. [Unsubscribe](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/scrollable_message_response.cc#L56)
2. [check to add scrollable message button subscribe](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/scrollable_message_response.cc#L56)

Everything works fine.

jacobkeeler, 2016-09-07:
@AnastasiyaBritanova Is there any requirement-based reason why [Alert buttons](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/alert_response.cc#L50) are not unsubscribed while [ScrollableMessage buttons](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/scrollable_message_response.cc#L55) are?

AnastasiyaBritanova, 2016-09-12:
@jacobkeeler SDL should stop processing UI.RPC-related SoftButtons notifications from HMI right after this UI.RPC is responded by HMI - it's true for both UI.Alert and UI.ScrollableMessage.

jacobkeeler, 2016-09-16:
@AnastasiyaBritanova From what I can tell, ScrollableMessage is currently the only RPC in which Core unsubscribes from the softbuttons when a response is received, should an issue be created for this?

joeygrover, 2016-09-19:
[image] QA - OnButtonPress not delivered</a>

madhuy, 2016-10-04:
@AnastasiyaBritanova Could you please your input on this issue?

AnastasiyaBritanova, 2016-10-04:
@madhuy, @jacobkeeler  i could comment from requirements point of view only: 

if a SoftButton was added via RPC other than `Show` (`Alert, ScrollableMessage, ShowConstatntTBT, AlertManeuver, UpdateTurnList`) 
-> SDL will allow notifications from such buttons only while the related RPC is active (= drop the notification after the related RPC is responded).

madhuy, 2016-10-04:
@AnastasiyaBritanova in that case, only scrollableMessage fallowing requirment I guess.

madhuy, 2016-10-12:
With 4.1 and integrating #660 this issue also fixed.

joeygrover, 2016-10-24:
Fixed with #909

closed 2016-10-24 by joeygrover