smartdevicelink/sdl_core #626: Generic DisplayType

MprenticeUIE, 2016-06-20:
Current DisplayType is based on Ford head units.  This needs to be changed to generic enums.  Example Video_Ht_Wt or Text_#lines_Lines. 

To this end, implement Video_800_480 included in the next SDL Core and Proxy Drop

MprenticeUIE, 2016-07-13:
quick clarification we want Video_480_800

goelgoldi, 2016-10-05:
@joeygrover, Any update on this request?

joeygrover, 2016-10-05:
This was discussed on a while back, but not finally decision has been reached. 

I believe an evolution proposal is needed here. I think the change will be more than adding an enum for a specific resolution. We might be able to add a generic enum like "SDL_Generic". Then I believe we want to add things like resolution/screen size/etc into the display capabilities. 

@tpulatha Can you comment? I know we talked about this. I think the quick answer is to add SDL_Generic as a Display_Type and then using the evolution process to add the other parameters.

goelgoldi, 2016-10-05:
@tpulatha Please let us know when its doable?

joeygrover, 2016-10-07:
@goelgoldi There is now a pull request open that should address the addition of the enum. #884

joeygrover, 2016-10-14:
Fixed with merge of #884

closed 2016-10-14 by joeygrover