smartdevicelink/sdl_core #627: Branch develop build failed: libusb.h No such file

BrandonHe, 2016-06-21:
build fail with develop branch, attached failed log:
```
[ 12%] Building CXX object src/components/transport_manager/CMakeFiles/TransportManager.dir/src/transport_manager_default.cc.o
In file included from /home/brandon/Brandon/brandonGitHub/test/sdl_core/src/components/transport_manager/include/transport_manager/usb/common.h:43:0,
                 from /home/brandon/Brandon/brandonGitHub/test/sdl_core/src/components/transport_manager/include/transport_manager/usb/usb_aoa_adapter.h:40,
                 from /home/brandon/Brandon/brandonGitHub/test/sdl_core/src/components/transport_manager/src/transport_manager_default.cc:45:
/home/brandon/Brandon/brandonGitHub/test/sdl_core/src/components/transport_manager/include/transport_manager/usb/libusb/usb_handler.h:43:20: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
make[2]: *** [src/components/transport_manager/CMakeFiles/TransportManager.dir/src/transport_manager_default.cc.o] Error 1
make[1]: *** [src/components/transport_manager/CMakeFiles/TransportManager.dir/all] Error 2
make: *** [all] Error 2
```

anosach-luxoft, 2016-06-21:
@BrandonHe `sudo apt-get install libusb-1.0`

BrandonHe, 2016-06-21:
@anosach-luxoft Thanks! As I build on Ubuntu, the issue is resolved. But I'm afraid remove libusb library from source code instead of installation by command may have an influence on port  transplantation. What's your opinion?

closed 2016-06-24 by BrandonHe